AllskyTeam/allsky-website #1: Keogram & Startrails index errors

jcauthen78, 2018-03-03:
The PHP is throwing errors with the "include_once" 

```
Warning: include_once(analyticstracking.php) [function.include-once]: failed to open stream: No such file or directory in /home/jimcauth/public_html/allsky/keograms/index.php on line 38

Warning: include_once() [function.include]: Failed opening 'analyticstracking.php' for inclusion (include_path='.:/opt/php/php-5.2.16-suhosin/lib/php') in /home/jimcauth/public_html/allsky/keograms/index.php on line 38
```
I don't know enough about PHP, but is the  **_include_once("analyticstracking.php");_**function good to just erase from both indexes?  
reference: [My AllSky Pages](http://jimcauthen.com/allsky/)

And on a related index.php note - the index of 'Videos' seems to have been thrown off a bit with the new CSS? it's not showing the smaller icons as it used to

thomasjacquin, 2018-03-03:
You can remove the include. I just added it to track visitors on my website.

Try grabbing a new copy of videos/index.php. Your page seems to slightly differ from mine.

jcauthen78, 2018-03-05:
I commented out "//" the lines and it fixed it up. 
Also updated the 'videos/index.php' and it looks correct now too.

closed 2018-03-05 by thomasjacquin