AllskyTeam/allsky-website #2: Installation allsky-website

Stewaro, 2018-09-19:
how do I instal allsky-website  on my Hoster? 
Or does she run only on the Raspberry?
Greetings.

thomasjacquin, 2018-09-19:
Hi,

You should be able to download the files and upload them to your hosting account. Nothing special about it. All you need is to change some settings (lat, long, image file name, etc). In order for the videos, startrails and keograms to be listed, you need your host to have PHP installed.

Thomas

Stewaro, 2018-09-22:
Hello Thomas,

Thank you very much.
Unfortunately I do not reach the goal.
  "All you need is to change some settings (lat, long, image file name, etc)."
PHP is running on my host. but error viewing videos, startrails and keograms. Internal Server Error
  - The server encountered an internal error or misconfiguration and was unable to complete your request. -
look at www.allskycam-rotheul.de
can you help?
Henry.

thomasjacquin, 2018-09-23:
Hi Henry,
It depends of the structure of your directories on your host.
I think you just need to remove `allsky` from the URL for each of those links.
Edit index.html lines 56, 57 and 58.

Stewaro, 2018-09-23:
Hi Thomas, 
it works now.
Thank you very much.

closed 2018-12-07 by Stewaro

Kilis71, 2019-03-05:
Hi, I have a problem on allskycam web, the photos are not loaded, the settings on raspberry are giiste but on the web remains the screen loading http://www.astrokilis.altervista.org/allskycam/

thomasjacquin, 2019-03-05:
Your host needs to have PHP running. Not all the hosts have it. Some of them only support html files.
Make sure you can use PHP files on altervista first.

Kilis71, 2019-03-05:
how do I know?

thomasjacquin, 2019-03-05:
It looks like their hosting plan has PHP capabilities. I'm not sure which one you are using.

Kilis71, 2019-03-05:
Which version should I use of php

Kilis71, 2019-03-05:
I can use versions 5.3, 5.3extra, 5.6, 5.7

thomasjacquin, 2019-03-05:
Any version should work but use 7 if you want, it's the latest one.

Kilis71, 2019-03-05:
the program on the server is working, what is not working and the image loading from raspberry to the server

thomasjacquin, 2019-03-05:
You need to fill your settings in `scripts/ftp-settings.sh`
You'll need your `host`, `username`, `password` and `directories`
You will need to create these directories on the host first.

Kilis71, 2019-03-06:
what you tell me I did but the photos do not load them. Does the domain need top-level or is it good even for the second level?

thomasjacquin, 2019-03-06:
What do you have in IMGDIR?
It should be the path to this directory on your server. I don't know how your server is set up. It could be `/var/www/html/public_html/allsky` or something like `lamp/vhosts/www.myserver/public_html/allsky`. It's different for every host.

Kilis71, 2019-03-06:
# FTP/SFTP settings
PROTOCOL='ftp'
USER='astrokilis'
PASSWORD='Stese16212'
HOST='astrokilis.altervista.org/'
IMGDIR='/allsky/'
MP4DIR='/allsky/videos/'
KEOGRAM_DIR='/allsky/keograms/'
STARTRAILS_DIR='/allsky/startrails/'

Kilis71, 2019-03-06:
http://www.astrokilis.altervista.org/allsky/

thomasjacquin, 2019-03-06:
Try these in Filezilla to make sure it's correct and it can connect and look at the directory structure on the host. Once you have all that, it should work fine.

Kilis71, 2019-03-06:
With filezilla working, I've uploaded allsky web folders and files. But from raspberry to the altervista host they do not load

thomasjacquin, 2019-03-06:
Try running the program manually in a terminal to see the error message:

service allsky stop
./allsky.sh

Kilis71, 2019-03-06:
Altervista uses self-signed certificates that cause me an error during loading, can I ignore the certified?

Kilis71, 2019-03-06:
It gives me a fatal certificate error

thomasjacquin, 2019-03-06:
Try adding this option to any lftp calls: `ssl:verify-certificate false`

It should look like that for example in saveImageDay.sh

`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; set ssl:verify-certificate false; put $FILENAME-resize.$EXTENSION; bye" &`

saveImageNight.sh and timelapse.sh have similar calls.

Kilis71, 2019-03-06:
I did as your instructions, now give me this error.

cd: Errore fatale: Certificate verification: certificate common name doesn't match requested host name ‘ftp.astrokilis.altervista.org’ (23:04:CA:15:B3:3B:D4:B1:A8:48:77:57:6E:22:58:9A:8C:81:33:D9)
put: image-resize.jpg: Errore fatale: max-retries exceeded (EPSV failed, will try PASV)

Kilis71, 2019-03-06:
if I remove ftp. from the host address from the allsky console, the connection remains in progress but does not connect. I'm going crazy

thomasjacquin, 2019-03-06:
Send me your credentials at jacquin.thomas@gmail.com and I'll give it a try on my side.

Kilis71, 2019-03-06:
do you want host server credentials?

thomasjacquin, 2019-03-06:
The ones you're using to connect with Filezilla.

Kilis71, 2019-03-06:
Email sent...

maphilli14, 2019-03-12:
This is quite intriguing and I'd like to set this up too.  Do you have a list of requirements, I see:

1) External website
2) PHP support
3) FTP file push?

what else?

thomasjacquin, 2019-03-12:
If you want to have your live image, keograms, startrails and timelapses available on the internet, you will need a web host that supports PHP. Most of them have PHP installed.
FTP is almost always available with your hosting account because that's how you set up a website.
That's all you need.