AllskyTeam/allsky-website #4: Constellation overlay and aurora activity not appearing.

CleverJay, 2019-03-14:
Hi Thomas.
I've got my allsky website up and running at https://www.skylapser.com/allsky/
The overlay and the aurora info are not appearing. I am not getting any errors in the Chrome javascript console and the network calls seem to have executed. I've also compared the mark-up to your site and can't see what would prevent the overlays on mine.
Thanks in advance for any insights.

thomasjacquin, 2019-03-14:
Hey Jay,

Do you know what version of PHP is installed on your server? I'm thinking that the aurora forecast is failing because getForecast.php returns an encoded response and the `json_encode` function is only supported on PHP 5.2 and higher.

CleverJay, 2019-03-14:
Thanks Thomas, The server is running PHP V5.6. that would have been too easy. :-) Do i have to set a document root in the functions.php?

thomasjacquin, 2019-03-14:
Can you try this on your website and see if it returns data?

`http://www.thomasjacquin.com/allsky/getForecast.php`

CleverJay, 2019-03-14:
Put your URL into controller.js Here:
`    $scope.getForecast = function () {

        function getSum(data, field) {
            var total = _.sumBy(data, function (row) {
                return parseInt(row[field]);
            });
            return Math.round(total / 7);
        }

        function getDay(number) {
            var day = moment().add(number, 'd');
            return moment(day).format("MMM") + " " + moment(day).format("DD");
        }

        $http.get("http://www.thomasjacquin.com/allsky/getForecast.php")
            .then(function (response) {
                $scope.forecast = {};
                $scope.forecast[getDay(0)] = getSum(response.data, "day1");
                $scope.forecast[getDay(1)] = getSum(response.data, "day2");
                $scope.forecast[getDay(2)] = getSum(response.data, "day3");
            });
    };

    $scope.getForecast();
}

`
And the forecast disappeared altogether.

CleverJay, 2019-03-14:
Got the following error in the JS console: angular.js:10661 Mixed Content: The page at 'https://www.skylapser.com/allsky/' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://www.thomasjacquin.com/allsky/getForecast.php'. This request has been blocked; the content must be served over HTTPS.

obviously it wants a local or secure source. I will try it as https if that doesn't work I will revert to just getForcast.php.

thomasjacquin, 2019-03-14:
Sorry, I meant try calling your own php file in the browser.

Mine returns this:
`[{"time":"00-03UT","day1":"0","day2":"3","day3":"2"},{"time":"03-06UT","day1":"1","day2":"3","day3":"2"},{"time":"06-09UT","day1":"2","day2":"2","day3":"2"},{"time":"09-12UT","day1":"1","day2":"2","day3":"2"},{"time":"12-15UT","day1":"2","day2":"3","day3":"2"},{"time":"15-18UT","day1":"2","day2":"3","day3":"2"},{"time":"18-21UT","day1":"3","day2":"3","day3":"2"},{"time":"21-00UT","day1":"3","day2":"4","day3":"3"}]`

Yours returns this:
`[{"time":"","day1":null,"day2":null,"day3":null}]`

I think there's a PHP function that fails to execute on your host but I'm not sure which one yet.

CleverJay, 2019-03-14:
Thanks Thomas. I will investigate further, There maybe something in the htaccess that is preventing php outside of wordpress. I am quite happy with just the camera for the time being. we are generally too far south for aurora, and the overlay is just icing on the cake.

thomasjacquin, 2019-03-14:
Actually, I think the error is not on `json_encode` but rather on `file_get_contents`. I believe it may be a permission issue.
My getForecast.php is set to 755 for permissions as well as the `allsky` directory

HPWever, 2019-06-02:
Hi all, I had the same issue. Found out (with the great help of Thomas) that it is due to …. the **language setting in 

virtualsky.json**

It has to be set to English because most other languages do not work!
        "lang": "en",

I have attached a word document describing the whole installation process.
[installing allsky-website.docx](https://github.com/thomasjacquin/allsky-website/files/3244648/installing.allsky-website.docx)

CleverJay, 2019-06-02:
Thank you so much  for the Word doc. I will try to reinstall the website
and hopefully won't lose my archive.

Best
--
Jay Shaffer
https://skylapser.com/allsky/

closed 2020-03-20 by CleverJay