AllskyTeam/allsky-website #6: FTP not uploading

ZebrasHaveBigEars, 2019-05-01:
Hi Thomas

I have faithfully followed the instructions (I hope!) but no files seem to be being transferred to my website. 

http://www.chrisjackson.biz/allsky/

I am just wondering if you can give me any advice to get operational? The website is windows based but it does run php and the actual software and the raspberry 3 on the site works ok.

Is there a log of errors to look at? 

Please note I am not an expert at any of this.

Thanks in advance for your support.

ZebrasHaveBigEars, 2019-05-02:
Using ./allsky.sh and setting the upload variables to true in the configuration file, I get 

"Resizing lsky/'  [Connecting ...]

"Uploading" 

"Saving ..."

With no reported error

Yet there is no upload of files to my website.

I am wondering if this might help with the diagnosis.

I have checked my website name, username and password with my ISP and they agree that they are correct.

Thanks!

ZebrasHaveBigEars, 2019-05-02:
I notice that after a while I get a bunch of 

[Delaying before reconnect: Number]

I assume this indicates some failure to connect?

thomasjacquin, 2019-05-03:
Hi,
I believe this is coming from the settings in `scripts/ftp-settings.sh`.
Can you show me the content of that file? You can mask the username and password. Seeing your settings would help me figure out the issue.
Thanks

ZebrasHaveBigEars, 2019-05-03:
Hi Thomas

Sure ... your assistance here is much appreciated. The contents of the file (with XXX representing masking) is here:

#!/bin/bash

# FTP/SFTP settings
PROTOCOL='ftp'
USER='XXX'         (includes special characters)
PASSWORD='XXX'   (includes special characters)
HOST='XXXX.XXXXX.XXXX.net.com'  
IMGDIR='/allsky/'
MP4DIR='/allsky/videos/'
KEOGRAM_DIR='/allsky/keograms/'
STARTRAILS_DIR='/allsky/startrails/'

ZebrasHaveBigEars, 2019-05-03:
I might know the problem ... I will check later

thomasjacquin, 2019-05-03:
I think the problem here is that the 4 paths need to be the full path on the server. For example /home/chrisjackson/public_html/allsky

You can see the full path when you connect to your directory using filezilla or other ftp software.

ZebrasHaveBigEars, 2019-05-04:
Thanks ... I am checking for the full paths with my ISP.  

Now I also get a fatal error which says "Certificate verification: certificate common name doesn't match the requested host name". I have tried adding the extra code "set ssl:verify-certificate no" into the  lftp line of code in the three files as suggested by you in another link but this does not work.

thomasjacquin, 2019-05-06:
Try changing the lftp calls and add this option `"set ssl:verify-certificate false; `

Such as 
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set ssl:verify-certificate false; set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye" & `

ZebrasHaveBigEars, 2019-05-08:
Hi Everyone

I have spent the last week grappling with the lftp command line as it has caused me no end of grief. I am making progress now and have developed test code plus split the command line into several different lines to more easily determine where errors arise. I just did my first successful ftp this morning using my new lftp command lines! Once I get my code fully working and tested I will post it so that people can work out their ftp problems more easily without having to ask Thomas all the time. One trick is to put "debug" into the lftp  command line so that you get proper errors reported when using ./allsky.sh to test how things are working.   Since "set ssl:verify-certificate false"  can lead to insecure connections (although it works well), I will test "set ssl:verify-certificate/ftp.website.com no" as a possible superior solution.

Thanks so much for such an exciting project! 

Cheers

ZebrasHaveBigEars, 2019-05-10:
ONLY IF YOU HAVE MAJOR FTP PROBLEMS, test  your ftp using this simple code saved into test.sh and then run from terminal on your pi from within the directory using ./test.sh.   Subcommand "debug" provides lots of help for errors and "dir" tells you the starting path for your allsky directory.

#!/bin/bash

lftp "
set ssl:verify-certificate no
set net:max-retries 1 
set net:timeout 20
open ftp://website.com
user 'username' 'password'
debug
cd
dir
lpwd
bye"
echo "FTP DONE"
date

In fact, I substituted this lftp code into the allsky scripts and simply added the transfer code provided by Thomas into it. After a week of trying to get the original code to work or using more complex code, this simple code was the only way I managed to get it to work and it was very easy to do.  The subcommand "set ssl:verify-certificate/ftp.website.com no" did not work for me for some reason.

It should be easy to substitute in the new code for Thomas' slicker code but which is harder to troubleshoot:

1. Make a copy of the original code
2. Put # in front of the original lftp command line
3. Once tested copy in the new lftp code below  the original lftp line (not the first line)
4.  Put in new code using "CD folder/folder" etc  to move to the correct folder in your remote website
5. Check Thomas' original lftp code and put in the appropriate PUT command into the script
6. Make sure you indent the new code to be positioned in the same place as the original
7. Make sure you don't forget the fi command!
8. Save.  Test using ./allsky.sh

If anyone wants examples then let me know.

HPWever, 2019-06-02:
I never got to work the FTP either. I now hapily use DropBoxUploader a
GitHub project by Andrea Fabrizi from Italy; this is how you install it:
git clone https://github.com/andreafabrizi/Dropbox-Uploader.git
Then give the execution permission to the script and run it:
 $ chmod +x dropbox_uploader.sh
 $ ./dropbox_uploader.sh
The first time you run dropbox_uploader, you'll be guided through a wizard in order to configure access to your Dropbox. This configuration will be stored in ~/.dropbox_uploader

ZebrasHaveBigEars, 2019-06-03:
Dropbox might work very well since it is presumably trying to update to the remote site all the time with the new files  whereas the FTP system only attempts once and gives up. I am finding ftp unreliable as a result especially as I am updating the day and night video combined which is about 80mb per day. Is the unreliability  the reason for using dropbox?

Can you give an example of the command to put in the script to get it to work on updating to a website (or does it not update to a website and only to a personal dropbox?). 

Thanks

ZebrasHaveBigEars, 2019-06-09:
I have now installed dropbox uploader.  It really is quite easy and works a treat. If anyone wants to see some example lines of code then let me know ... it would save a few hours of messing about.

elineberry75, 2019-10-27:
I would love to see some examples of how you are using the dropbox uploader.  FTP is giving me a fit!

HPWever, 2019-12-08:
Sorry for my delay in answering.
After installing dropboxuploader as described above...
Just add the following lines to the file "allsky/scripts/timelapse.sh" on your Pi:

if [ "$UPLOAD_VIDEO" = true ] ; then
# next line is added to upload to DROPBOX using the dropbox_uploader.sh
       /home/pi/Dropbox-Uploader/dropbox_uploader.sh upload /home/pi/allsky/images/$1/allsky-$1.mp4 /
fi

closed 2021-09-23 by EricClaeys