AllskyTeam/allsky-website #9: Not really an issue but...

bnred777, 2019-08-01:
Just wondering if it is possible to use the full image scale on my website for live view? If so, how would I go about doing that? Also, is it possible to stretch the image for a larger view on live view (on my website), similar to going full screen in the GUI live view?

By the way, this project and software is great. Thank you so much for the work put into this.

http://astrocandy.org/allsky/

That's a new site and all I have right now is the allsky cam setup.

Thanks,
B

thomasjacquin, 2019-08-01:
Hi!

Yes you can upload the full size image to your website. 
At the bottom (the lftp line) of the scripts saveImageDay.sh and saveImageNight.sh just replace this

 $FILENAME-resize.$EXTENSION

By this 

 $FILENAME.$EXTENSION

That should do it.

bnred777, 2019-08-01:
Great! Any way of changing the image container size on the website?

bnred777, 2019-08-01:
By the way, you can add me to the map if you'd like

thomasjacquin, 2019-08-01:
You should be able to make the container larger by modifying the allsky.css file.

There’s a few hardcoded values with 960px. You can use the number you want instead.

Line 90 controls the width of the image:
max-width: 960px;

I’ll add you to the map when i am back home (currently visiting family in Europe) and I have access to my computer.

closed 2021-09-23 by EricClaeys