AllskyTeam/allsky-website #11: PHP Errors and other problems

WirthmU, 2019-09-03:
I am having some troubles with the website - I did a fresh install but still the same problems are occurring:

The startup screen keeps showing the „Loading ...“ jpg file. I do have a current image-resize.jpg that is updated regularly by the camera.

The constellations overlay only shows a globe grid without any constellations overlaid on the loading.jpg file

Archived Keograms show the following error message:
Warning: mkdir(): File exists in /volume1/web/Saanenmoeser/allsky/keograms/index.php on line 29 Call Stack: 0.0000 245064 1. {main}() /volume1/web/Saanenmoeser/allsky/keograms/index.php:0 0.0002 246536 2. mkdir() /volume1/web/Saanenmoeser/allsky/keograms/index.php:29

Archived Startrails show the following error message:
Warning: mkdir(): File exists in /volume1/web/Saanenmoeser/allsky/startrails/index.php on line 29 Call Stack: 0.0000 245176 1. {main}() /volume1/web/Saanenmoeser/allsky/startrails/index.php:0 0.0002 246648 2. mkdir() /volume1/web/Saanenmoeser/allsky/startrails/index.php:29

Any help to this?

WirthmU, 2019-09-03:
I have changed the name of the file in the config.js to the right one - that handled the startup screen problem.

thomasjacquin, 2019-09-03:
Make sure you have the GD library enabled on your Synology NAS.

WirthmU, 2019-09-05:
I have checked with Synology support and they claim that GD is enabled. I could also check it using the php56 -i command -its there.

I noticed that the problem only exists for keograms and startrails, the timelapses are looking normal.
Any other ideas

thomasjacquin, 2019-09-05:
The reason I was suggesting Checking the GD library is because another user had the exact same issue with startrails and keograms and activating GD worked for him.

https://stackoverflow.com/questions/26356268/call-to-undefined-function-imagecreatefromjpeg-and-gd-enabled

I sent him this link and that did the trick for him:

> In Synology packages screen, I went to Web Server. In PHP section, I enabled GD.

thomasjacquin, 2019-09-05:
Try commenting out `mkdir('thumbnails', 0755);` in both startrails/index.php and keograms/index.php

That will remove the warning from your logs at least.

WirthmU, 2019-09-09:
Commenting out as you suggested corrected the warnings.
Everything is working now - thank you for your support.

closed 2019-09-09 by WirthmU