AllskyTeam/allsky-website #14: Instructions on how setup allsky-website

wannaberocker, 2019-10-13:
Hello, I would like to get this all allsky-website, I clones into github and retrieved the code but I am not sure what to do next or how to see my own allsky cameras allsky-website. What do I have to do to get this working? Thanks.

wannaberocker, 2019-10-13:
I seem to have it working now but the image isn't being uploaded how can I do this?

wannaberocker, 2019-10-13:
On other words, what do do for imageName: ...??

wannaberocker, 2019-10-14:
Another weird quirk....Internet Explorer browser instantly reflects my changes to config.js and controller.js files while Chrome does not. Why is that? I'm just trying to set the overlay up correctly and I am totally lost on how to do this. Any help would be appreciated Thanks.

wannaberocker, 2019-10-14:
This is so incredibly difficult to what should be an almost mindless adjustment, I'd like to offer money but at this point its been so frustrating I feel like I should be getting paid LOL! Anyway I've gotta take a step back as I have been screwing with this for 5 days straight now.

thomasjacquin, 2019-10-15:
Hello, can you share the url? I will take a look. It can be caused by various reasons. Sometimes the hosting account doesn't have PHP installed, sometimes it's a permission issue on the directories.

wannaberocker, 2019-10-15:
Here is the URL, also how do you adjust the sky overlay to fit properly? I seem to be chasing my tail on that. Thanks.

http://wannaberocker.ddns.net:10700/allsky-website/

thomasjacquin, 2019-10-15:
It looks like I can't access the URL but from what I can see, you've installed the website on the RPi. I think it should work as well but originally this project was designed to be installed onto a hosting account.
That should still function properly though. I have accessed the public page here (http://wannaberocker.ddns.net:10700/public.php) For the constellation overlay, it covers a 180 degrees field of view so that might not fit your current field of view. It looks like your lens doesn't cover the whole sky. You may need a different type of projection. See the projection examples on the VirtualSky project page (http://slowe.github.io/VirtualSky/).

wannaberocker, 2019-10-15:
I am unsure why you cannot access the URL as I have had a few other people able to access it. Is there a way to get the timelapse, keograms, and star trails to appear with the server installed on the RPi? I’m just looking to do that for now if possible. Thanks.

maserowik, 2020-06-06:
> I have since figured out how to fix my broken links on the allsky-website page. It was simply going to the index.html file and editing the href for each item. That was easy. One problem though remains...the startrails and keograms, although routed correctly now does not display any of the jpgs saved in the directory the link points to. Why is this? FYI the url is now http://rko.ddns.net:10700/allsky-website/

It appears that you are using you raspberry pi to locally host the allsky-website.  Would you be kind enough to share how you got it running?  

Thanks
Mike

Torakin9366, 2020-07-30:
Hallo zusammen. Ich möchte auch gerne die Webseite einrichten. Habe die zip Datei auf meinem Raspberry pi 4 herunter geladen und entpackt, der Ordner liegt noch im Download Ordner. Meine Daten hab ich in der config.js schon geändert. Was muss ich jetzt weiteres tun, bin absoluter Neuling auf dem Gebiet. Meine Allsky Kamera läuft und die Daten werden auf meinem Ftp Server gespeichert.
Danke schon mal für die Bemühung
Torsten

klingonio, 2020-08-17:
Hallo

> Hallo zusammen. Ich möchte auch gerne die Webseite einrichten. Habe die zip Datei auf meinem Raspberry pi 4 herunter geladen und entpackt, der Ordner liegt noch im Download Ordner. Meine Daten hab [...]

Hi Torsten,
an dem Punkt bin ich auch. Schau mal in dem Thread https://github.com/thomasjacquin/allsky-website/issues/7
Das sollte Dein Problem lösen. Ich versuche mich da auch mal .... melde mich.

wannaberocker, 2020-08-17:
> It appears that you are using you raspberry pi to locally host the allsky-website. Would you be kind enough to share how you got it running?
>
> Thanks
> Mike

Hi Mike, sorry I did not see this until today. I can remember mostly what I did, I installed the allsky-website onto the var/www/html directory on a raspberry pi 4. I had to edit some code to get the timelapse, startrails, and keograms to be archived and viewed properly. I even use dropbox-uploader via the endOfNight script to upload the high quality stills to use to make a higher quality timelapse if I wanted to do so.

klingonio, 2020-08-19:
Hi wannaberocker
what code did you edit?
i also installed the allsky-website on my pi under /var/www/html and had the same problems (HTML Error 404 ...)

wannaberocker, 2020-08-19:
Try to get everything to work locally first. I can’t exactly remember what all I did. I saved all my edited files. I’ll see if I can get them to you.

huisinr, 2020-12-10:
Howdy !
With regards to the allsky-website portal  and  archived videos, is it possible to list the newest to the oldest in descending order?
http://www.huisinga.org/allsky/videos/
 Thanks in advanced for any suggestions!

thomasjacquin, 2020-12-10:
See this https://github.com/thomasjacquin/allsky-website/issues/25#issuecomment-700912841

PhotonPup, 2021-07-17:
I am struggling with this. If anyone could assist, start to finish getting the AllSky-Website onto a Pi4 and functioning that would be amazing.

Shand359, 2021-09-10:
i have the same issue as PhotonPup.... I have no issues now getting the gui installed and running ok but no matter what i do i cannot get the website to work.....

PhotonPup, 2021-09-10:
> i have the same issue as PhotonPup.... I have no issues now getting the gui installed and running ok but no matter what i do i cannot get the website to work.....

Hey Shand359 - I was able to get up and running thanks to the help of someone from the AllSky FB page. Because of that I put together a guide i guess you  could say in case anyone else had same issues i did. You can check what I put in place as a walk through...maybe it will help. 
https://www.photonpup.com/programming-allsky-software-onto-pi4

EricClaeys, 2021-09-23:
@wannaberocker, can this issue be closed?  I believe an answer and a like were provided.

closed 2021-10-02 by EricClaeys