AllskyTeam/allsky-website #15: Centering and Sizing the Virtual Sky Overlay

wannaberocker, 2019-10-19:
What is the proper way to get the Virtual Sky Overlay to set correctly on the allsky image? It seems the settings in config.js that are text will work but anything having to do with the offset does not change anything, same for controller.js. Any adjustment seems to not have any effect except for changing the text displayed.

FredoTheGreat, 2019-10-30:
I would also like to know this, it appears that cache has an influence on the position because i have made changes that only appeared to work the next day.
i have seen some peoples allsky  web pages looking perfectly aligned with the star chart so i would like to replicate.

WirthmU, 2019-11-20:
I can adjust the Virtual Sky Overlay position, the vertical and horizontal placement and the size adjusting config.js and virtualsky.json. Depending on the browser the changes can bee seen immediately or only the next day. The constellation names and stars are only seen if the browser language is set to english.

The problem I have is that the overlay is leaking out over the bottom border of the camera image. Is there a way to adjust the overlay boundaries?

brianboru82, 2020-02-17:
I'm also curious about this, if there is any adjustment for the field of view of the fisheye lens, like a 150 degree versus a 180 degree...

nday91, 2020-04-24:
To center virtual sky, you can change the "projection" in virtualsky.js to be Lambert and then edit the overlaySize and offset parameters in config.js.  It takes a little trial and error.  Reload the page every time you make a change.

Here are the docs for the values in virtualsky.js https://slowe.github.io/VirtualSky/

svego, 2021-02-24:
Did you ever got it working correctly? If so could you share the config files?

FredoTheGreat, 2021-02-24:
i no longer have mine after running it for a year the sunlight destroyed the camera so i moved onto another project
one thing I found was you must clear the browser cache everytie you make a change otherwise you will not see a change

EricClaeys, 2021-09-23:
Closing issue.  If you have the issue of your changes not showing up until the next day, then in your browser, press F12 to get into developer mode.  On the upper right, look for the "Network" tab and open it.  Then check the "disable cache" box (might be called something different on your browser).  As long as you have the F12 window open, it will re-read the config.js file every time you refresh the window.

closed 2021-09-23 by EricClaeys