AllskyTeam/allsky-website #16: Image not displaying, only loading...

uzelessknowledge, 2019-12-28:
I have the site stood up on my host. After finally getting the FTP to upload my issue is now the image does not display. I only see the loading screen. When I look at the index.html i see
<div id="live_container">
[image]

Should this image be pointing to the loading.jpg? I did not see this when I inspected your site. I changed the src to be the name of the image. Was this the correct thing to do?

thomasjacquin, 2019-12-31:
The `loading.jpg` image should stay at the root. It's included in the repository. It is used as a placeholder while the website is reading the allsky image.
Make sure you have the correct image name in `config.js`
Instructions are here: https://github.com/thomasjacquin/allsky-website

uzelessknowledge, 2019-12-31:
Yes, I understand that the loading.jpg should be at the root. All I see is the loading...
If I change the index.html file to point to the image then it loads. 

config.js
`var config = {
    title: "Alabaster, AL",
    imageName: "image-resize.jpg",
    location: "Alabaster",
    latitude: 33.249,
    longitude: -86.807,
    az: 0,
    camera: "ASI120MC",
    computer: "Raspberry Pi 3",
    owner: "Travis Rice",
    auroraMap: "north",
    overlayOffsetLeft: 0,
    overlayOffsetTop: 0,
}`

Index.html that does not work
`<img id="current" class="current" src="loading.jpg">`

Index.html that does work
`<img id="current" class="current" src="images/image-resize.jpg">`

site: [https://funrices.net/allsky/](url)

thomasjacquin, 2019-12-31:
It will work if `image-resize.jpg` is at the root. It seems that yours is in an `images` directory.

uzelessknowledge, 2020-01-01:
Thanks. That worked.

closed 2020-01-01 by uzelessknowledge