AllskyTeam/allsky-website #17: Instruction for setting up the Raspberry for Website upload

onkelgeorg, 2020-02-11:
Hi all,

may be I have missed something but I can not find any information of how to setup the Raspberry for file upload. 

The website is running here: http://www.torhoff.com/allsky/ but obviously shows no images.

So my question is: Which settings are required in the Raspberry - or how do I put the images on my website?

Cheers,
Joerg

thomasjacquin, 2020-02-11:
Hi Joerg,

The instructions are in the README of the other repo: https://github.com/thomasjacquin/allsky
Basically, you just need to make sure that you have set the UPLOAD_IMG variable to true in `config.sh` and set your FTP credentials in `scripts/ftp-settings.sh`

Thomas

onkelgeorg, 2020-02-11:
Hi Thomas,

thanks for the quick reply. I made the changes you suggested but it still doesn't work for some reason. I will check it in depth tomorrow.

Cheers,
Joerg

thomasjacquin, 2020-02-11:
It varies with each hosting company but for me it didn't work the first time because I was trying to use SFTP and I never used SSH on that host before.
So I did the following `ssh allsky.myhost.com` and it asked me if I wanted to add myhost.com to the list of know hosts.
After that if worked fine.
Maybe your problem is different. You can start the program manually by running `./allsky.sh` to get the debug information. It should tell you why the upload is failing.

onkelgeorg, 2020-02-12:
Great idea to start it manually and check the debug information.
Currently I see 2 issues:

1. 
_Exposure value: 8392500 µs
Saving...20200212 06:56:23
Auto Gain value: 10
Auto Exposure value: 8392 ms
Sleeping: 11618 ms
**/home/pi/allsky/config.sh: Zeile 1: true#!/bin/bash: Datei oder Verzeichnis nich                                                  t gefunden
/home/pi/allsky/scripts/ftp-settings.sh: Zeile 1: username#!/bin/bash: Datei ode                                                  r Verzeichnis nicht gefunden**_

So it seems that there is a problem with the config.sh and the ftp-settings.sh

2.
_Uploading

cd: Fataler Fehler: Certificate verification: certificate common name doesn't ma                                                  tch requested host name »xyz.com« (long number in brackets)
put: image-resize.jpg: Fataler Fehler: Certificate verification: certificate com                                                  mon name doesn't match requested host name »xyz.com« (long number in brackets)_

Here is a certificate problem whatever it means. Actually I am uploading data from a different (WIndows) PC with exactly the same credentials used on the Raspberry. That leaves me a little bit lost...

onkelgeorg, 2020-02-12:
Ok it seems that I solve problem #1: I removed "true" in front of #!bin bash in the config.sh and did the same with "username" in ftp-settings.sh
So the error message disappeared.

Regarding problem #2 I found hints that adding set ssl:verify-certificate false somewhere might help. But where to add it and first of all: isn't it a security risk?

thomasjacquin, 2020-02-12:
You'll need to add it where lftp is used: saveImageDay.sh, saveImageNight.sh, timelapse.sh, etc

This is how to use it: 	
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; ssl:verify-certificate false; put $FILENAME-resize.$EXTENSION; bye" &`

onkelgeorg, 2020-02-12:
Thanks again, and sorry to bother you again. 

I followed your advice and added **set ssl:verify-certificate false;** like you suggested, but the problem persists. 

I also tried set ssl:verify-certificate no; and set ftp:ssl-allow no; but nothing worked :(

thomasjacquin, 2020-02-12:
Have you tried `ssh torhoff.com`? to add it to your list of known hosts?

onkelgeorg, 2020-02-13:
No I haven't tried that. 
I tried to do it upon your suggestion, but I don't know how and where to add. The information I found on Google is pretty irritating. Any help is appreciated.

onkelgeorg, 2020-02-14:
Ok, I got it running. The solution was to use the servers host name.

Thanks for your help and patience

closed 2020-02-14 by onkelgeorg