AllskyTeam/allsky-website #18: Updating virtual sky

pnoah, 2020-02-16:
I seem to have posted this to the wrong forum.

I downloaded and installed VirtualSky 0.7.4 on my server. When Allsky tries to access it, nothing happens. No sky map is produced. Has anyone succeeded in updating to the latest version?
Thanks
Paul

thomasjacquin, 2020-02-17:
Hi Paul, on your browser, open the developer tools and look for the console tab. There are probably some red messages here.
I remember trying to update a while back and got some conflict with the stripped jQuery version used by the new version of Virtual Sky.

pnoah, 2020-02-21:
Thank you that helped.  I had to make a change in controller.js from planetarium = $.virtualsky(data);
to planetarium = S.virtualsky(data);
and add these lines to index.html
    <script src="virtualsky2/stuquery.js"></script>
    <script src="virtualsky2/virtualsky.js" type="text/javascript"></script>
    <script src="virtualsky2/virtualsky-planets.js"></script>

thanks for the help

closed 2020-02-21 by pnoah

thomasjacquin, 2020-02-22:
Ah! Thanks for the info, I'll give it a try on my side.