AllskyTeam/allsky-website #19: Sunset time not correct

beorn59, 2020-02-25:
Hi,
I came across this issue because virtualsky is sometimes showing constellations and sometimes just a plain grid. I see the constellations during daytime and the grid at night. I think this should be the other way round. Also the console log prints that it is night-time during my daytime and vice-versa.

My first thought was, that I have to adjust the timezone from Yukon to Germany, but I did'nt find a setting for this.

Long story short, I tried to read the code and it compares the current time, to the time of sunset. The sunset-time is read from file data.json, which has a timestamp from last night, but the content shows 
{"sunset": "2016-05-27T00:24:00.000-0200"}
which is some years ago. 
I cant see where this content comes from. But maybe someone has an idea.

kind regards
Beorn

thomasjacquin, 2020-02-25:
Hi Beorn, Can you post your website address? I'll take a look.

beorn59, 2020-02-25:
Hello Thomas,

the camera is still under construction and not publicly available yet. I send you the url by PM.
kind regards
Beorn

closed 2021-09-23 by EricClaeys