AllskyTeam/allsky-website #20: All Sky image is not loading onto webpage. Loading image only presented.

spacetrace1982, 2020-04-11:
Good Evening. I hope that you can help. 

I have set up the website and the All sky cam is now sending images to my host via FTP. However, I am still seeing the Loading page on the website. 

On the FTP server: 
The Image "image-resize.jpg"is saving into a directory one level up from my public_html folder. 

For Information the config.js is as below 

var config = {
    title: "The Black Cat Observatory",
    imageName: "/image-resize.jpg",   _#Just a note - I have tried with and without the "/"_
    location: "North Wales, Penrhyn Bay",
    latitude: 53.1,
    longitude: -003.05,
    az: 180,
    camera: "ASI20M",
    computer: "Raspberry Pi 3",
    owner: "Tracey Snelus",
    auroraMap: "north",
    overlaySize: 875,
    overlayOffsetLeft: 0,
    overlayOffsetTop: 0,

The Index files are pretty much untouched 

<html prefix="og: http://ogp.me/ns#">
<head>
    <title>BlackCatAllSky</title>
    <meta property="og:title" content="The Black Cat Observatory AllSky" />
    <meta property="og:type" content="image/jpeg" />
    <meta property="og:url" content="http://www.thomasjacquin/allsky/" />
    <meta property="og:image" content="http://www.thomasjacquin.com/allsky/image.jpg" />
	<link rel="shortcut icon" type="image/png" href="allsky-favicon.png">
    <link rel="stylesheet" href="https://maxcdn.bootstrapcdn.com/font-awesome/4.5.0/css/font-awesome.min.css">
    <link rel="stylesheet" href="https://maxcdn.bootstrapcdn.com/bootstrap/3.3.6/css/bootstrap.min.css"
          integrity="sha384-1q8mTJOASx8j1Au+a5WDVnPi2lkFfwwEAa8hDDdjZlpLegxhjVME1fgjWPGmkzs7" crossorigin="anonymous">
    <link href='https://fonts.googleapis.com/css?family=Ubuntu:400,300' rel='stylesheet' type='text/css'>
    <link rel="stylesheet" type="text/css" href="animate.min.css">
    <link rel="stylesheet" type="text/css" href="allsky.css">
    <link rel="stylesheet" type="text/css" href="allsky-font.css">

    <script src="https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js"></script>
    <script src="https://maxcdn.bootstrapcdn.com/bootstrap/3.3.6/js/bootstrap.min.js"
            integrity="sha384-0mSbJDEHialfmuBBQP6A4Qrprq5OVfW37PRR3j5ELqxss1yVqOtnepnHVP9aJ7xS"
            crossorigin="anonymous"></script>
    <script src="https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js"></script>
    <script src="https://cdnjs.cloudflare.com/ajax/libs/moment.js/2.11.2/moment.min.js"></script>
    
    <script src="virtualsky/virtualsky.js" type="text/javascript"></script>
    <script src="ng-lodash.min.js"></script>
    <script src="config.js"></script>
    <script src="controller.js"></script>
</head>
<body ng-app="allsky" ng-controller="AppCtrl">
	<div class="header">
		<div class=title>The Black Cat Observatory All Sky</div>
		<div ng-show="forecast" class="forecast pull-right">
			<span>Aurora activity: </span>
			<span class="forecast-day" ng-repeat="(key,val) in forecast">{{key}}:
				<span ng-class="getScale(val)"
					  title="{{val}}/9">{{getScale(val)}}</span> </span>
		</div>
		<div style="clear:both;"></div>
	</div>
	<div class="info animated slideInRight" ng-show="showInfo==true">
		<ul>
			<li><i class="fa fa-fw fa-map"></i>Location: <span>North Wales, Penrhyn Bay</span></li>
			<li><i class="fa fa-fw fa-map-marker"></i>Latitude: <span>53.18N</span></li>
			<li><i class="fa fa-fw fa-map-marker"></i>Longitude: <span>003.05W</span></li>
			<li><i class="fa fa-fw fa-camera"></i>Camera: <span>ZWO ASI 120M</span></li>
			<li><i class="fa fa-fw fa-gears"></i>Computer: <span>RaspberryPi</span></li>
			<li><i class="fa fa-fw fa-user"></i>Owner: <span>J&THarty</span></li>
		</ul>
	</div>
	<span class="notification" compile="notification"></span>

	<ul id="sidebar" class="animated slideInLeft">
		<li><i class="fa fa-2x fa-fw allsky-constellation" id="overlayBtn" title="Show constellations overlay" ng-click="toggleOverlay()" ng-class="{'active': showOverlay}"></i></li>
		<li><a href="../allsky/videos" title="Archived Timelapses"><i class="fa fa-2x fa-fw fa-play-circle"></i></a></li>
		<li><a href="../allsky/keograms" title="Archived Keograms"><i class="fa fa-2x fa-fw fa-barcode"></i></a></li>
		<li><a href="../allsky/startrails" title="Archived Startrails"><i class="fa fa-2x fa-fw fa-circle-o-notch"></i></a></li>
		<li><i class="fa fa-2x fa-fw fa-camera" title="Information about the camera" ng-click="toggleInfo()" ng-class="{'active': showInfo}" style="margin-top: 5px; font-size: 1.9em"></i></li>	
	</ul>

	<div id="imageContainer">
		<div id="starmap_container" ng-show="showOverlay==true">
			<div id="starmap"></div>
		</div>
		<div id="live_container">
			[image]
		</div>
	</div>
	
	<div class="diy"><a href="http://thomasjacquin.com/make-your-own-allsky-camera"><i class="fa fa-gear"></i> Make Your Own</div>

	<script src="analyticsTracking.js"></script>
</body>
</html>

the website if you wish to see and inspect it can be found at: 
https://muciferous-invoices.000webhostapp.com/

I am happy to send FTP svr settings if necessary. I am certain that it is going to be a simple fix and I am missing something rather simple. 

Thank you 

Tracey

thomasjacquin, 2020-04-11:
Good evening Tracey,

Try this in `config.js`.
```
longitude: -3.05,
```

spacetrace1982, 2020-04-11:
Hello Thomas

 

Thank  you for the suggestion. I have made the change and the website is now giving me the aurora forecast successfully. However today I have had to shift everything onto a new host server. I set it all up again to the point where the image-resize.jpg is being loaded to the htdocs folder of the website, however still no success with it loading onto the webpage. It is still displaying to loading image. 

Thank you 

 

 

www.blackcatobservatoryallskycam.co.uk <http://www.blackcatobservatoryallskycam.co.uk>

spacetrace1982, 2020-04-11:
the correct URL is http://www.blackcatobservatoryallsky.co.uk/

thomasjacquin, 2020-04-11:
There doesn't seem to be an image at this address: http://www.blackcatobservatoryallsky.co.uk/current/image-current.jpg

Make sure you RPi is uploading successfully. Since you've changed servers, you might need to add the new IP to the known hosts. Try `ssh blackcatobservatoryallsky.co.uk`. It'll probably ask if you want to add this website to your know hosts.

spacetrace1982, 2020-04-11:
Good Evening 

 

So first I just tried to get theimages to upload to /htdocs/ so that I know the images are loading to the host. Then I can repoint the webpage to the right images location. 

 

I updated the ftp config file and double checked everything is ok. 

I ran allsky .sh so I could see what was happening and I see the following 

 

 

aving...20200411 23:15:27

Auto Exposure value: 12196 ms

Sleeping: 7805 ms

Resizing 

 

Uploading 

 

Exposure value: 12196548 Âµs.]

Saving...20200411 23:15:39

Auto Exposure value: 12196 ms

Sleeping: 7805 ms

Resizing docs' [Not connected]  

 

Uploading 

 

Exposure value: 12196548 Âµs.]  

Saving...20200411 23:15:51

Auto Exposure value: 12196 ms

Sleeping: 7805 ms

Resizing 

 

Uploading

 

Sleeping: 7805 ms

Resizing docs' [Delaying before reconnect: 22]  

 

Uploading ocs' [Delaying before reconnect: 3]   

 

Exposure value: 12196548 Âµsore reconnect: 11]  

Saving...20200411 23:18:18

Auto Exposure value: 12196 ms

Sleeping: 7805 ms

Resizing docs' [Delaying before reconnect: 10]  

 

Uploading ocs' [Delaying before reconnect: 3]   

 

Exposure value: 12196548 Âµsore reconnect: 63]  

Saving...20200411 23:18:30

Auto Exposure value: 12196 ms

Sleeping: 7805 ms

Resizing docs' [Delaying before reconnect: 34]  

 

Uploading ocs' [Delaying before reconnect: 33]  

 

Exposure value: 12196548 Âµsore reconnect: 51]  

Saving...20200411 23:18:42

Auto Exposure value: 12196 ms

Sleeping: 7805 ms

Resizing docs' [Delaying before reconnect: 50]  

 

Uploading ocs' [Delaying before reconnect: 33]  

 

 

I presume that this is not loading to the host

spacetrace1982, 2020-04-11:
I have managed to sort it all out now. Thank you for your patience and help 

 

Tracey

closed 2021-09-23 by EricClaeys