AllskyTeam/allsky-website #22: New website up and running mostly. Few issues.

BrightskyObservatory, 2020-07-28:
Hello,

I have my Allsky camera setup at my remote location and have a website setup for it, but the Keogram and Startrail images are not updating and I get an error on those pages, could you please try help me on how I can fix this error. If you click on the link to the Keogram and Startrail you will see what I mean.

UR:: http://brightskyobservatory.bsch.com.au/allsky

Andrew.

thomasjacquin, 2020-07-28:
Hi Andrew, looks like a permission issue. You probably need to set permissions to 755 on these 2 directories. See if that helps https://stackoverflow.com/questions/27648431/imagejpeg-unable-to-open-file-for-writing-permission-denied

BrightskyObservatory, 2020-07-30:
Hi Thomas,

I did have the permissions set to 755 so not sure what is happening, one thought is, I have the images coming in as .png files not jpeg files, I wonder if this has something to do with it? I am not an expert or even a beginner in coding so I have no idea, it was just a thought since the extensions are different. I noticed this when I saw that the 4 images that were there, although with broken thumbnail images, had jpeg extensions, this may have been before I swapped to png. Still doesn't explain the file permissions error  though.

Andrew

maserowik, 2020-08-06:
Andrew  try this it appears to solve my issues
https://github.com/thomasjacquin/allsky/issues/74

BrightskyObservatory, 2020-08-06:
Managed to get everything working. I needed to change the permissions of the two directories (Startrails and Keograms) to 777 for it to work correctly. Once I did that, everything started working properly. Now trying to add a info (i) button to the animated Left Sidebar so I can add a info page about the Building of the Allsky camera.

maserowik, 2020-08-06:
There should already be a icon in the lower right corner that says _Make Your Own_

BrightskyObservatory, 2020-08-06:
@maserowik Yeah there is, but I like to try and reverse engineer things, I was able to work out the code for the sidebar and add my own icons to it, so I can add the info about my camera build. Pretty easy once you know the code.

closed 2021-09-23 by EricClaeys