AllskyTeam/allsky-website #23: Trying to enable the do not display image during the day.

Dentonknifeworks, 2020-08-14:
What do I need to be changed and or correct for turning off the day time image at the web site. I would like it to only show evening images and during the day the count down to dark.

I see in controller.js that it has the code to do this and data.json has a option set to true/false
but not been able to get it to work so I am thinking I am missing a setting. 
Thanks
Brad
site is www.dentonknifeworks.com/allsky

thomasjacquin, 2020-08-14:
Hi Brad, 

To achieve that, you need to:

- Set POST_END_OF_NIGHT_DATA to true in `config.js`.
- Update postData.sh to your location by changing these variables (I need to fix this to remove the hardcoded values)
latitude=60.7N
longitude=135.02W
timezone=-0700

Then you should start seeing a `data.json` file at the root of your website at the end of each night.
To force it the first time, just call `scripts/postData.sh`

Dentonknifeworks, 2020-08-14:
Thanks Worked like as it was intended.
Brad.

closed 2020-08-16 by Dentonknifeworks