AllskyTeam/allsky-website #24: VirtualSky: how to get objects on the map

DarKOrange75, 2020-08-18:
Hi,
How can I get objects like M42 to display on the VirtualSky map?

Thank you,

Laurent

thomasjacquin, 2020-08-18:
You can add objects to this file https://github.com/thomasjacquin/allsky-website/blob/master/virtualsky.json

DarKOrange75, 2020-08-19:
Thank you Thomas but it does not work (I have emptied the cache before testing). I can't find any piece of code that could read and handle this variable.

DarKOrange75, 2020-08-19:
Ok I've found the issue.

I am running an HTTPS web site and the code in virtualsky.js was trying to server content from 2 insecure http servers.
I had to modify lines 2930, 2931 and 1161 to switch from http to https. Now it works!

			$.ajax({ dataType: "jsonp", url: 'https://www.strudel.org.uk/lookUP/json/?name='+ob[o], context: this, success: function(data){

			url = p.url || "http**s**://server1.wikisky.org/v2?ra="+(p.ra/15)+"&de="+(p.dec)+"&zoom=6&img_source=DSS2";

			img = p.img || 'http**s**://server7.sky-map.org/imgcut?survey=DSS2&w=128&h=128&ra='+(p.ra/15)+'&de='+p.dec+'&angle=0.25&output=PNG';

closed 2020-08-19 by DarKOrange75