AllskyTeam/allsky-website #25: Latest videos, startrails and keograms first

DarKOrange75, 2020-09-29:
Hi @thomasjacquin,

I see on your AllSky web site that the latest video, startrails and keograms are displayed first. How do you achieve that? On my web site the oldest is displayed first and the latest is last. Which is fine for now because my setup is new but in a few months it may be uncomfortable.

Thank you!

Laurent

thomasjacquin, 2020-09-29:
Change this line https://github.com/thomasjacquin/allsky-website/blob/master/videos/index.php#L25 and do it for keogram and startrails as well.

DarKOrange75, 2020-09-29:
Thank you @thomasjacquin :-)

closed 2020-09-29 by DarKOrange75

Kilis71, 2024-08-30:
> Change this line https://github.com/thomasjacquin/allsky-website/blob/master/videos/index.php#L25 and do it for keogram and startrails as well.
> 
> Replace with `rsort($files);`

I tried but this modification doesn't work for me