AllskyTeam/allsky-website #26: It's not dark yet message with aurora image instead of camera image

martinmagnan, 2020-10-02:
Hi
since about a week I get a message saying It's not dark yet in Shawinigan. Come back in x hours. I Have not change a thing since installation and it worked before as expected.

I have search a bit in the files but my skills are a bit limited

here is a screen capture

Any idea for a solution ?

Thanks

[image]

martinmagnan, 2020-10-02:
I found the solution here 
https://github.com/thomasjacquin/allsky-website/issues/23#issuecomment-674167523

thomasjacquin, 2020-10-08:
Hi Martin, do you have a URL to your allsky page? I can add you to the map if you want.

martinmagnan, 2020-10-08:
Hi Thomas
it's not yet on a public server. I'll send you a link when it is.

martinmagnan, 2021-01-13:
Hi @thomasjacquin 

the website is https://allsky.astropatate.com/allsky/

Martin

closed 2021-09-23 by EricClaeys

EricClaeys, 2022-01-05:
@martinmagnan Where did you get the diagram of the Earth with the Aurora forecast over it?  Someone asked if we can display it again.

Thanks - Eric

martinmagnan, 2022-01-06:
@EricClaeys When I first installed the Allsky website this image was displayed during the day because my location was not properly set. My allsky camera is running since september 12th 2020 with very few interruption and I have not touch it since.  I remember that I de-activated the prediction image to get the cam image displayed instead. The image is available here https://services.swpc.noaa.gov/images/animations/ovation/north/latest.jpg but I dont remember how to display it on the allsky website.

EricClaeys, 2022-01-06:
@martinmagnan Thanks for the info.  Once I updated the data.json file on the website and turned off daytime capture and turned on aurora display, the world image appeared.  The data.json file that was on the website never got updated and had a sunset time of 2016, so the current time was never before the sunset time and hence the website never thought it was daytime so didn't display the world image.

I am working on a fix.