AllskyTeam/allsky-website #29: I can't see the constellations

abrosio, 2020-11-24:
Hi Thomas and hi everyone,
I can't see the constellations but only the lines, why?
I tried all the settings but also nothing in the virtualsky.json file...
Many Thanks
Antonino

thomasjacquin, 2020-11-24:
Are you saying `virtualsky.json` is empty on your website?
It should contain these lines: https://github.com/thomasjacquin/allsky-website/blob/master/virtualsky.json

abrosio, 2020-11-25:
Yes the files is ok but only the grid lines are seems.
This is my site: http://abobservatory.altervista.org/allsky/

Can you add my all sky camera to the map? Thanks

thomasjacquin, 2020-11-30:
Hi Antonio,

Did you figure out the issue? It seems to work when I look at your website. The overlay doesn't really line up with your image because of cropping I assume but the constellations are here. I'll add you to the map!

[image]

abrosio, 2020-12-02:
It is really weird because my PCs, smartphones and macs only see the orange lines and not the constellations
[image]

thomasjacquin, 2020-12-02:
I could be a language issue. I suppose your OS and browser is set to Italian. Try to switch temporarily to English. If it works, then I'll have to look into bringing multi language support.

abrosio, 2020-12-02:
Great Idea Thomas, you are absolutely right, i changed the language and now it works!
how to solve?  I have seen from your map other cams in Italy which also works with Italian browsers.

abrosio, 2020-12-02:
I found the problem!  you have to add it.json file in virtualsky / lang and then it works
Thanks

abrosio, 2020-12-03:
Then you have to definitely act on one of the files, could you help me figure out which one?
Thanks

abrosio, 2020-12-17:
Hello Thomas,
Did you then find the solution to be able to view the constellations in other languages too? I was not able to solve by myself unfortunately

Thanks

thomasjacquin, 2020-12-23:
You'd probably need to add the language here as well https://github.com/thomasjacquin/allsky-website/blob/master/virtualsky/virtualsky.js#L788

HamOP, 2020-12-26:
You can simply pull the latest version. I suggested adding `"lang": "en"` to the virtualsky.json and Thomas merged it into the official version. This will make it work independently of the site user's language setting - the only drawback is that the virtual sky names are always English which is not such a big deal in my opinion. Issue is that language is chosen according to the viewer's system language and as long as you don't have any possible language inside there, there will always be page visitors where it will not work.

closed 2021-09-23 by EricClaeys