AllskyTeam/allsky-website #30: Snow Removal

maserowik, 2020-12-02:
This is not and issue with the program it is with the outside environment.

 I have my allsky-website up and running.  I live in Pittsburgh PA area and we got our first snow.  Looking at the image does anyone have a good way to get the snow off the dome?

http://n3bsq.com/allsky/

Thanks

jcauthen78, 2020-12-02:
I've been researching the same thing for a frost problem on my end - one idea was to get a USB lens warmer strip and and somehow affix it to the top inside of the case so that the heat would rise up towards the dome. My only hesitation is how to turn it on & off via the Pi, as well as i haven't researched the amp draw it wants either.

HamOP, 2020-12-26:
I added six 100 Ohms resistors in parallel around my lens and switch them on and off with a shell script and a simple relay board. Outside temperature is read by a DS18b20 sensor. Works perfectly. The heater draws about 0,2 amps and is easily fed by my 5V power supply for the RPi. I can provide you with some details if you want.

maserowik, 2020-12-26:
Can you provide you scripts

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

HamOP, 2020-12-26:
No problem, script is below. You need to activate the 1wire-bus for the DS18b20 and replace the serial number in the second line with the one of your sensor.
Relay ist wired to GPIO 18. The script reads outside temperature and switches on the heater if it's below 3°C (3000). If higher than 5°C, heater is switched off. It also writes the temperature to the allsky temperature.txt file for dark subtraction as the Pi HQ camera doesn't provide the temperature. You can disable that if you want (lines 5 and 6).
Second function of the script is to switch the fan of my RPi case depending on CPU temperature over GPIO 14.

The script needs pigpio installed to work:
`sudo apt-get install pigpio`
`sudo systemctl start pigpiod`

Then, add the script to your crontab to execute e.g. every minute.
`sudo crontab -e`

Mine looks like this:
```
*/1 * * * * /home/pi/temptest.sh
@reboot echo 0 >/sys/class/leds/led0/brightness
@reboot echo 0 >/sys/class/leds/led1/brightness
00 12 * * * sudo reboot
```
First line is the heater script running every minute, lines 2 and 3 switch off the RPi LEDs on boot and line 4 reboots the RPi every day at 12 am.

Hope it helps!

Script temptest.sh:
```
#!/bin/bash
TEMP=$(sed -n "s/^.*t=//p" /sys/bus/w1/devices/28-0316a7ad9aff/w1_slave)
SCALE=1000

rm /home/pi/allsky/temperature.txt
echo $((TEMP / SCALE)) >> /home/pi/allsky/temperature.txt

if (( TEMP > 5000 ))
	then
	pigs w 18 0
fi

if ((TEMP < 3000))
	then
	pigs w 18 1
fi

TEMPCPU=$(cat /sys/class/thermal/thermal_zone0/temp)

if (( TEMPCPU > 45000 ))
        then
        pigs w 14 1
fi

if ((TEMPCPU < 42000))
        then
        pigs w 14 0
fi
```

closed 2021-09-23 by EricClaeys