AllskyTeam/allsky-website #32: Using Startrails thumbnails for videos

CleverJay, 2020-12-14:
What do I need to change in videos/index.php to use the startrail image as the thumbnail for the videos?

thomasjacquin, 2020-12-16:
Replace `src="../aurora_snap.jpg"` on this line https://github.com/thomasjacquin/allsky-website/blob/master/videos/index.php#L39 by this `src="./thumbnails/startrails-".$date.".jpg"`

Not tested but that's the idea.

CleverJay, 2020-12-18:
Thanks Thomas

CleverJay, 2020-12-18:
Had to make the following changes but now it works: `if (file_exists("../startrails/thumbnails/startrails-".$date.".jpg")){
				//echo "[image]";
				echo "<a href='./$file'><div class='day-container'><div class='image-container'>[image]</div><div>$year-$month-$day</div></div></a>";
			} else {
				echo "<a href='./$file'><div class='day-container'><div class='image-container'>[image]</div><div>$year-$month-$day</div></div></a>";
			}`

closed 2020-12-18 by CleverJay