AllskyTeam/allsky-website #34: Timelapses, keograms and startrails in reverse order

HamOP, 2020-12-26:
I just saw that on my install of the website, all of the nightly files are ordered "oldest one first", while in the demo website and all the other ones I looked at, order is reversed and the last night is shown first.
It was easy to fix by replacing the `asort()` by `arsort()`, but I don't understand why this happens on my site. Could that have something to to do with PHP versions?

AstroHighlander, 2020-12-28:
Hi HamOP - where do I find the asort( ) or arsort( ) commands(s)?  I want to fix this as well. Eric

HamOP, 2020-12-28:
In the `index.php` files in the videos, keograms and star trails folders. Look for the line `asort($files);`.

AstroHighlander, 2020-12-29:
HamOP thanks very much for that - it worked a treat!  Sooooo easy when you know how

EricClaeys, 2021-08-19:
HamOP,
If this issue is resolved, would you please close it?
Thanks - Eric

closed 2021-08-19 by HamOP

Kilis71, 2024-09-01:
> Nei `index.php`file nelle cartelle video, keogrammi e scie stellari. Cerca la riga `asort($files);`.

in the latest version this line of code is missing

EricClaeys, 2024-09-01:
@Kilis71, 
Most of the code in the the `index.php` files in the keograms, startrails, and videos directories moved to a common location in the current release, to make maintenance easier.  The `asort($files)` command is now in the `functions.php` file.

Are you having a problem with sorting?

Kilis71, 2024-09-03:
ok, thank you very much