AllskyTeam/allsky-website #35: Want to increase image size on website liveview

AstroHighlander, 2020-12-29:
The liveview image size on my AllSky Admin Panel is perfect but when I view it on my own website it is too small (I have to zoom the view to 150%-175% to get it to the size I want).

Hopefully there is an easy way to fix this.  Eric

AstroHighlander, 2021-01-13:
I would still welcome some help with this - I just can't figure out how to sort it.  The image resolution when I zoom it up isn't that great either.

svego, 2021-01-21:
I would also like to know this

EricClaeys, 2021-09-23:
@AstroHighlander , @svego 
Closing - I added to the "to do" list to give people instructions or somehow make it easier to change the size.
Changing the image size isn't very hard - it's getting the overlay to match that is difficult.

You can try playing with the "overlaySize" in /var/www/html/allsky/config.js to change the size of the star overlay, and the "width" and "height" in /var/www/html/allsky/virtualsky.json.
In /var/www/html/allsky/allsky.css, play with "max-width:" near the top.  There might be other settings in there that need changing as well.
It's not easy

closed 2021-09-23 by EricClaeys