AllskyTeam/allsky-website #36: Issue witj transfering images from Pi to ftp

Gary79, 2021-02-08:
Hi Thomas, can you help I am struggling to get the Pi to send anything to the ftb host.
Website is all set up and I am recieving all to the Pi interface but just keep getting a loading image on the website. 

Here's the URL allsky.salfordastro.org.uk 

Cheers 
Gary

horatti, 2021-02-08:
Hi Gary!

Here is an example if you use Thomas' original files:

lftp "ftp"://"username":"$PASSWORD"@"allsky.salfordastro.org.uk":"/subfolder (if any..)" -e "set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye"
..... or so if a certificate problem occurs:
lftp "ftp"://"username":"$PASSWORD"@"allsky.salfordastro.org.uk":"/subfolder (if any..)" -e "set ftp:ssl-allow off; set net:max-retries 1; set ssl:verify-certificate false; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye"

Try it on the command line first (without "bye"), then paste it into the appropriate files.

Ati

Gary79, 2021-02-09:
Thanks Ati All sorted now

rseg80, 2021-05-15:
> Hi Gary!
>
> Here is an example if you use Thomas' original files:
>
> lftp "ftp"://"username":"$PASSWORD"@"allsky.salfordastro.org.uk":"/subfolder (if any..)" -e "set net:max-retries 1; set net:timeout [...]

Where does this code go?  I’m having the same issue.  Thanks

horatti, 2021-05-15:
Hi!

In any file that uploads somewhere on ftp. For example:
endOfNight.sh
saveImageDay.sh
saveImageNight.sh
uploadForDay.sh

The files are in the "scripts" subdirectory.

EricClaeys, 2021-08-19:
Gary,
If this issue is resolved, and it sounds like it is, would you please close it?
Thanks - Eric

closed 2021-09-23 by EricClaeys