AllskyTeam/allsky-website #37: Allsky and webserver on the same device

HB665054, 2021-02-22:
Hi, I am very happy about your nice software, but I have a small problem

My hardware is Raspberry 3+ and RPiHQ Camera

I installed the website software,
I edited the config.js and I have my own website.
But, there is no live image :-(
Also, the public.php is empty:-(

How can I transfer the live image to the website?

Do I need  ftp when the images and the webserver are on the same device?

Thank you in advance!
Henry
 

Best regards

Heinrich

scorot, 2021-04-11:
Hi,

in order to run the Allsky-wesite and the Allsky service on the same device without setting up an ftp server, I ended up in tunning the script file endOfNight_additionalSteps.sh to create symlinks to the files (videos, startrails, etc...). See my code here after.
I also installed the Allsky admin web GUI, so I did not need to install apache since lighthttpd is allready running for that. I simply put the file of the allsky-website into /var/www/html/allsky/. After that, the script creates the symlinks in  /var/www/html/allsky/ pointing to /home/pi/allsky/.

```
#!/bin/bash

#
# Place any additional code you require to be run at the end of the night in this script. This script is run prior
# to the deletion of any old image files.
#

source $ALLSKY_HOME/config.sh
source $ALLSKY_HOME/scripts/filename.sh

if [ $# -eq 1 ] ; then
	if [ "x$1" = "x-h" ] ; then
		echo "Usage: $BASH_ARGV0 [YYYYmmdd]"
		exit
	else
		LAST_NIGHT=$1
	fi
else
	LAST_NIGHT=$(date -d '12 hours ago' +'%Y%m%d')
fi

if [[ $KEOGRAM == "true" ]]; then
    OUTPUT="$ALLSKY_HOME/images/$LAST_NIGHT/keogram/keogram-$LAST_NIGHT.$EXTENSION"
    ln -s $OUTPUT /var/www/html/allsky/keograms/keogram-$LAST_NIGHT.$EXTENSION
    #echo -e "\n"
fi

if [[ $STARTRAILS == "true" ]]; then
    OUTPUT="$ALLSKY_HOME/images/$LAST_NIGHT/startrails/startrails-$LAST_NIGHT.$EXTENSION"
    ln -s $OUTPUT /var/www/html/allsky/startrails/startrails-$LAST_NIGHT.$EXTENSION
    #echo -e "\n"
fi

if [[ $TIMELAPSE == "true" ]]; then
   ln -s $ALLSKY_HOME/images/$LAST_NIGHT/allsky-$LAST_NIGHT.mp4 /var/www/html/allsky/videos/allsky-$LAST_NIGHT.mp4
fi
```
Cheers,
Sebastien

thomasjacquin, 2021-04-12:
I just pushed an update to the allsky repository that handles hosting the website on the RPi. 
See documentation here https://github.com/thomasjacquin/allsky#show-your-sky-on-your-own-website
You'll have to re-install the GUI first in order to fix permissions.

loloderu, 2021-04-19:
Hi,
Thank you so much for your work!
I am also hosting the website on the RPI and timelaspse page is not working properly, showing this (I downloaded the website tools on th 14 april 2021, so just after your update). Do yo u have any idea? 
[image]

loloderu, 2021-04-19:
I add the information that all video are existing in the directory images/LAST_NIGHT/*.mp4

thomasjacquin, 2021-04-23:
I added a script to install the website locally and updated the README

loloderu, 2021-04-23:
I found my error: I just forgotten to install php on my rpi!

HB665054, 2021-04-29:
> 
> 
> I added a script to install the website locally and updated the README

Thank you very much, it is working now!!!!
I have only one small issue: I cannot display the keograms, startrails, videos .... on the website
It looks the same as loloderu  shows above

php is installed!

EricClaeys, 2021-08-19:
HB665054,
if this issue is resolved, please close it.
Thanks - Eric