AllskyTeam/allsky-website #38: thumbnails for videos and images

rhine59, 2021-03-03:
Morning Thomas. Notice that you have thumbnails in your startrails .... directory but that the git clone install of your site does not. Can you please share how you did this. I am guessing that you used something like http://fancyapps.com/fancybox/3/ but as it is now a commercial product that you have removed it from your distro.

Any suggestions please? If possible I would like to avoid fees here, but if needs must .... ;-)

Many thanks, Richard

thomasjacquin, 2021-03-03:
Hi Richard,

You don't need any other product to get thumbnails. It's part of the code. It looks for a thumbnail image and if it can't find one, it will generate one.
https://github.com/thomasjacquin/allsky-website/blob/master/startrails/index.php#L38
Maybe your server failed to create the thumbnails directory? It should be located inside the startrails directory and have 755 permissions.

rhine59, 2021-03-04:
Many thanks Thomas. User error - changed startrails, keograms and videos subdirectories to 755 and all is good with the world once more. Richard

EricClaeys, 2021-08-19:
Richard,
If this issue is resolved (and it sounds like it is), would you please close it?
Thanks - Eric

EricClaeys, 2021-09-23:
Closing - no reply from user in over 1 month.

closed 2021-09-23 by EricClaeys