AllskyTeam/allsky-website #40: Gain problem on day images

flipper07, 2021-03-14:
Hello I have a problem with the daytime images.
I set Auto exposure and Auto gain to Yes.
I do not see how to improve.
I have another problem it is for my website that I just added your site.
We can see live, but the link for Timelapse / Keograms / startrails does not work.
I do not know how to do.
To be honest, this is the first time that I have created a site.
I don't think there is much to do, but I can't see.
Thanks for your help.

thomasjacquin, 2021-03-16:
Gain is always 0 by day. (c.f. doc here https://github.com/thomasjacquin/allsky#configuration)
I don't know if Orange offers PHP hosting. It seems that pagesperso is probably just plain HTML.

EricClaeys, 2021-08-19:
flipper07,
Your server doesn't have PHP on it, which is why you are seeing the PHP code (for example:  "; foreach ($files as $file)..." and why you can't view the startrails, keograms, and timelapse.

Since this issue isn't related to the allsky-website application, would you please close the issue?
Thanks - Eric

EricClaeys, 2021-09-23:
Closing - no response from user in over a month, and issue appears to have been resolved.

closed 2021-09-23 by EricClaeys