AllskyTeam/allsky-website #41: Messier Objects on overlay not showing

wramey7274, 2021-03-16:
Greetings:

The Messier Objects are not showing up on the overlay.  I have included below my virtualsky.json file. If anyone could help me identify the issue, I would certainly appreciate it.

virtualsky.json ~

{
	"id":"starmap",
	"projection":"fisheye",
	"width": 900,
	"height": 972,
	"constellations": true,
	"mouse": false,
	"gridlines_eq": true,
	"keyboard": false,
	"showdate": false,
	"showposition": false,
	"constellationlabels": true,
	"sky_gradient": false,
	"gradient": false,
	"transparent": true,
	"objects": "M1;M4;M7;M8;M11;M13;M16;M27;M31;M42;M45;M51;M57;M81;M82;M101",
	"lang": "en"
}

Thank you.

thomasjacquin, 2021-03-16:
Hi William, try replacing `http` by `https` in this file https://github.com/thomasjacquin/allsky-website/blob/master/virtualsky/virtualsky.js#L1161

I believe the browser is giving an error when loading the Messier objects because they are pulled from an http source and your website is using the https protocol. I haven't tried it but I think it'll work.

wramey7274, 2021-03-16:
Thomas,

 

You are a genius! Of course that was the issue. When I was looking at the code for virtualsky/virtualsky.js last week, I never considered that was the issue, even though when I code I make sure that the https is part of the URL, for the server will not render an external source without it.

 

Again, you are a genius! Thank you very much for your very rapid reply.

 

I should receive my additional hardware tomorrow or Thursday, and then I can pitch my allsky cam properly to achieve the overlay much more exact to the live-view than I have it at this time.

 

Thank you for your time invested, your skill, and your broad knowledge of troubleshooting. You certainly deserve another “cup of coffee” on me.

 

Bill

 

Bill Ramey

Email:  ***@***.***> ***@***.***

Cell: 404-285-9850

leitmori, 2021-04-08:
Unfortunately the LookUP service has been suspended: https://www.strudel.org.uk/lookUP/

leitmori, 2021-04-12:
Fixed with PR #42

closed 2021-09-23 by EricClaeys