AllskyTeam/allsky-website #43: Italian language issue with it.json

johnhardening, 2021-04-12:
Hi, setting the language in Italian ("lang": "it") in the virtualsky.json file only the outline of the Milky Way is shown.
I tried both generating a new it.json by editing en.json and fr.json and also getting the file from here https://raw.githubusercontent.com/slowe/VirtualSky/gh-pages/lang/it.json
Editing virtualsky.json and / or changing inside the translation file "code": "it" is enough to block the display.
If I take the it.json file and rename it to en.json, changing "code" to "en" and setting ("lang": "en") in the virtualsky.json file everything works by displaying the boundaries of the constellations, the planets and the names in Italian.

johnhardening, 2021-04-12:
Sorry, solved.
Added the line " 'it': {" language ": {" name ":" Italiano "," alignment ":" left "}}," in row 795 to the virtualsky.js file.
Thank you

EricClaeys, 2021-08-19:
John,
Sounds like the issue is resolved.  If so, please close it.
Thanks - Eric

closed 2021-08-19 by johnhardening