AllskyTeam/allsky-website #45: No videos on website

shiner501, 2021-06-16:
Hi,
Loving the project and will deffo buy you some coffees Thomas :). I have two questions please?

****Q1*****
Installed Allsky and the website on Pi 4B.  All cool.  
http://sjk.zapto.org:1289/allsky

I can see time-lapse videos on admin page just fine and yet *not* on the website.  I check /var/www/html/allsky/videos and no videos in there.  However, there are Keogram and Stratrails in those respective folders in /var/www/html/allsky.  Timelapse videos *ARE* in /allsky/images/ddmmyy just fine hence why it works on admin page.  But they are not in the /var/www/html/allsky/videos.

In ftpsettings.sh I have:

PROTOCOL='local'
USER='admin'
PASSWORD='secret'
HOST='example.com'
IMGDIR='/var/www/html/allsky/'
MP4DIR='/var/www/html/allsky/videos'
KEOGRAM_DIR='/var/www/html/allsky/keograms'
STARTRAILS_DIR='/var/www/html/allsky/startrails/'

And in config.sh I have set up correctly too:
# Choose between ZWO or RPiHQ or auto (prefers ZWO if present)
CAMERA="ZWO"

# Set to true to upload current image to your website
UPLOAD_IMG=true

# Set to true to upload timelapse video to your website at the end of each night
UPLOAD_VIDEO=true

# Set to true to send data to your server at the end of each night
POST_END_OF_NIGHT_DATA=true

# Set to true to generate a timelapse at the end of the night
TIMELAPSE=true

# Set the width and height parameter for timelapse generation (sizes must be divisible by 2)
TIMELAPSEWIDTH=1280
TIMELAPSEHEIGHT=960

# Timelapse frame rate (number of frames per second)
FPS=17

# Set to true to generate a keogram at the end of the night (image summary of the night)
KEOGRAM=true

# Set to true to upload the keogram after it has been generated
UPLOAD_KEOGRAM=true
<snip>

Why is videos OK on admin page but not visible in website?

*****Q2*****

Startrails does not look like Startrail.  Maybe it is because too bright since it is summer and not dark long at my latitude and I need to tinker with the darkness parameter?

Many thanks!
Steve

shiner501, 2021-06-17:
No worries, I worked it out and now get timelapse videos.
I used Sebastian's code in post [here](https://github.com/thomasjacquin/allsky-website/issues/37)

I also changed permissions of videos folder in to 777.

TBH I am not 100% sure which of these fixed it.  I will find out for the benefit of others and come back.
Steve

EricClaeys, 2021-08-19:
Shiner501,
Did you update your allsky-website files?  Thomas posted some updates that should have fixed the problem.
Either way, if this issue is resolved, please close it.
Thanks - Eric

EricClaeys, 2021-09-23:
Closing - no response from user in over a month, and user said issue was resolved.

closed 2021-09-23 by EricClaeys