AllskyTeam/allsky-website #46: Need to flip virtual sky overlay around North is South and South is North Currently

PhotonPup, 2021-07-24:
Default view for virtual sky is showing North is South and South is North in my current view. Is there a way to flip this around?

EricClaeys, 2021-07-29:
Try modifying the "az" variable in the config.json file.  az=0 puts North at the bottom of the picture.
I believe this is mentioned in the README document.

PhotonPup, 2021-07-29:
Hello EricClaeys,
Thank you for the information. One issue I am running into with the config.js is nothing in there is shown on the AllSky view so I changed in-line the index.html file which worked. 
Ill give this a shot again however and wait a day or so to see if this takes.

PhotonPup, 2021-07-29:
I left the editing in the index.html file and swapped just az = 0 in config.js. All good. thank you.

EricClaeys, 2021-08-19:
PhotonPup,
If this issue is resolved, please close it.
Thanks - Eric

PhotonPup, 2021-08-19:
Closing issue.

closed 2021-08-19 by PhotonPup