AllskyTeam/allsky-website #53: Isssue #50 No Video Thumbnails created

maserowik, 2021-09-29:
I have read the issue listed above about no thumbnail being created.  I read the readme but I still do not understand why i am  getting the no thumbnail on my video's .  I have a thumbnail on my startrails and keograms.

[image]

EricClaeys, 2021-10-01:
@maserowik this is on the website (with the constellation overlay), not the GUI, right?
Look in /var/log/lighttpd/error.log to see if it gives a clue.
If you click on the "No Thumbnail" picture, does it show the video?
Do you have /var/www/html/allsky/videos/thumbnails directory?  If not, try creating it.

maserowik, 2021-10-01:
Correct on the website NOT on the gui

Lighttpd error log

i@Pi-AllSky:~ $ tail -f /var/log/lighttpd/error.log
2021-09-30 11:35:59: (mod_fastcgi.c.421) FastCGI-stderr: deleting image directory: rm -rf /home/pi/allsky/images/20210929/
2021-09-30 11:36:18: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: delay in /var/www/html/includes/liveview.php on line 10
2021-09-30 11:36:39: (chunk.c.831) file shrunk: /home/pi/allsky/liveview-image.jpg
2021-09-30 11:36:39: (connections.c.456) connection closed: write failed on fd 11
2021-09-30 14:14:34: (chunk.c.831) file shrunk: /home/pi/allsky/liveview-image.jpg
2021-09-30 14:14:34: (connections.c.456) connection closed: write failed on fd 8
2021-10-01 00:47:42: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: PHP_AUTH_USER in /var/www/html/includes/authenticate.php on line 2
2021-10-01 00:47:42: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: PHP_AUTH_PW in /var/www/html/includes/authenticate.php on line 3
2021-10-01 05:03:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: PHP_AUTH_USER in /var/www/html/includes/authenticate.php on line 2
2021-10-01 05:03:08: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: PHP_AUTH_PW in /var/www/html/includes/authenticate.php on line 3

when I click on the No Thumbnails i get a video

here is that the hosting site shows.
[image]

Thanks 
Mike

EricClaeys, 2021-10-02:
@maserowik Thanks for the info.  Are the videos, startrails, and keogram files on your Pi in /var/www/html/allsky, or are they on your hosting server?

The thumbnails should be created automatically when you visit the page with the thumbnails on it.
The video thumbnail is created using ffmpeg, so if your files are on the hosting server, check that it has ffmpeg; if not, you won't get thumbnails.
Our "to do" list has the ability to create thumbnails on the Pi and upload them to remote hosts.  In your case, this would eliminate the need to have ffmpeg on the remote host.

maserowik, 2021-10-02:
@EricClaeys 

I have the website on a hosting company.

something did change  i use to get all the files startrail/keogram and the timelapse.mp4 left on my pi.  since the latest upgrade non of them are left on my pi.  

I do not recall having to place the ffmpeg on my hosting company site before.

I think I may have multiple issue currently.  not sure how to troubleshoot.  any suggestions?

website is 
[www.n3bsq.com/allsky](url)

Mike

EricClaeys, 2021-10-02:
@maserowik Mike,
Version 0.7 used a static aurora icon for the time lapse thumbnails, so ffmpeg wasn't needed.   Version 0.8 attempts to create the time lapse thumbnails automatically (just like version 0.7 did with startrails and keograms), and if that fails it displays the new "No thumbnail" image.  So nothing changed with time lapse thumbnails other than the default thumbnail it displays.
You could replace the "No thumbnail" image with anything you want until we implement the ability to create the thumbnail on the Pi and upload it.

Regarding not leaving files on your Pi, where did it used to leave them?
They should at least be in ~/allsky/images/DATE.  Those are the files that are uploaded to your hosting server and they are only deleted if AUTO_DELETE is "true" in config.sh and they are at least NIGHTS_TO_KEEP nights old.

Please attach (don't copy/paste) your config.sh and ftp_settings.sh (after having erased the username and password), and I'll take a look.

Eric

linuxkidd, 2021-10-02:
I wonder if it makes sense to instead use a startrails thumbnail?  This would eliminate the ffmpeg need, and still be roughly representative of the timelapse it links to.

EricClaeys, 2021-10-02:
@linuxkidd Michael, creating the thumbnail on the Pi and uploading to the host shouldn't be too difficult, and it's been a requested feature.  Not all hosting platforms support the commands/libraries needed to create thumbnails.

maserowik, 2021-10-02:
i just did a new git pull let me see if this works now.
I should know in about 14 hours.

Mike

EricClaeys, 2021-10-12:
@maserowik Any update?

WirthmU, 2022-07-13:
Any news to this issue?

I got the same error - no thumbnails generated in the videos/thumbnails directory:

[image]