AllskyTeam/allsky-website #61: functions.php:  on line 120 Warning: file_get_contents(ALLSKY_CONFIG/config.sh): failed to open stream: No such file

AndreasLMeg, 2021-11-05:
@EricClaeys,
am trying to get the current version of allsky-website up and running.

With the startrails I got the following error message:

`Warning: Use of undefined constant ALLSKY_CONFIG - assumed 'ALLSKY_CONFIG' (this will throw an Error in a future version of PHP) in /home/.sites/258/site4180562/web/astro/allsky2/functions.php on line 120 Warning: file_get_contents(ALLSKY_CONFIG/config.sh): failed to open stream: No such file or directory in /home/.sites/258/site4180562/web/astro/allsky2/functions.php on line 12 
`
https://www.astro-fotografie.at/allsky2/startrails/

Is this line responsible for it?
`$ thumbnailSizeX = get_variable (ALLSKY_CONFIG. '/ config.sh', 'THUMBNAILSIZE_X =', '100');`

I have no config.sh on the ftp server ...
Unfortunately I am not a HTML / PHP expert, so I need your help

AndreasLMeg, 2021-11-05:
btw: Are there any test devices (ZWO / Raspi HQ) including Allsky WEB page with the current software?

linuxkidd, 2021-11-06:
That line is definitely the cause.  It seems that `ALLSKY_CONFIG` and the `config.sh` references got added to the website code by accident ( or ahead of the current rest of the code state ).  You can replace the `get_variable....` part, and set the thumbnailSizeX manually in code.  It'll look like:

```
$thumbnailSizeX = 100;
```

This should allow the site to work normally until we can get an updated code merge to resolve it in the repo.

closed 2021-11-06 by ckuethe

AndreasLMeg, 2021-11-06:
> That line is definitely the cause. It seems that `ALLSKY_CONFIG` and the `config.sh` references got added to the website code by accident ( or ahead of the current rest of the code state ). You can [...]

@linuxkidd,
Thx, quickfix is working.

EricClaeys, 2021-11-06:
@AndreasLMeg you can also just define the missing variable:
`define(ALLSKY_CONFIG, '/home/pi/allsky/config');`

That is the ultimate solution for allsky-website on a Pi.

AndreasLMeg, 2021-11-06:
@EricClaeys  would be this a solution for pi and other ftp server ?
```
	if (isset($_ENV["ALLSKY_CONFIG"]))
          $thumbnailSizeX = get_variable(ALLSKY_CONFIG .'/config.sh', 'THUMBNAILSIZE_X=', '100');
        else  
          $thumbnailSizeX = 100;
```

AndreasLMeg, 2021-11-06:
btw: there ist another mistake in the file - please use $thumbnailSizeX instead of thumbnailSizeX (without"$")

PR #62 (2021-11-06): Revert "Use THUMBNAILSIZE_X instead of hard-coded number"
Reverts thomasjacquin/allsky-website#59

Website should not depend on `config.sh` since there's no guarantee it'll be present.

Resolves #61