AllskyTeam/allsky-website #68: aurora image

maserowik, 2022-01-05:
Is there a way to bring back the Aurora Iage?

https://github.com/thomasjacquin/allsky-website/issues/26

Mike

maserowik, 2022-01-06:
@EricClaeys 
Understand that there is a fix for this.  Thanks for you assistance on this Eric.

Mike

EricClaeys, 2022-01-07:
@maserowik I am working on a permanent solution.  In the meantime, you just have to update the data.json file in the same directory as the config.js file.  Assuming the data.json file in ~/allsky/tmp is updated as part of the endOfNight.sh script you can simply copy the file.
You also need to make sure you have the display auroras option turned on in config.js and that you are NOT taking daytime images.

maserowik, 2022-01-11:
@EricClaeys 
So i update my config.js and added the following
[image]

the data.json is appears to be correct for me.
[image]

daytime capture and daytime save are set to false.
[image]

but on the website the still seeing the image
[image]

any idea of what step I am missing?

Mike

EricClaeys, 2022-01-11:
@maserowik Mike, during the day when the image is showing, press F12 and on the right side click on the "Console" link.  You may need to first click on >>> to display the Console link.
Let me know what it says.  It might be something like "It's  nighttime".

maserowik, 2022-01-11:
@EricClaeys 
here is the website f12 setting
[image]

here is from the admin console
[image]

Mike

maserowik, 2022-01-12:
I went back to showing the live camera image.

EricClaeys, 2022-01-16:
@maserowik There is a new "controller.js" file that will hopfully show the aurora image during the day when you aren’t taking pictures.   The algorithm to determine daytime didn't always work.
Let me know how it goes.

EricClaeys, 2022-01-18:
Closing - issue fixed.  The old website code thought it was nighttime even when it was daytime so it never displayed the aurora image.
The new version also has a much more obvious "Come back at xxx" message - it's larger, brighter, and includes the time.

closed 2022-01-18 by EricClaeys