AllskyTeam/allsky-website #74: Aurora Forecast

maserowik, 2022-01-18:
@EricClaeys 
so in order for me to get the Aurora Forecast I had to do the following
I currently have my image save to a folder on my website called 
/domains/n3bsq.com/public_html/allsky/image/
[image]

with the data.json and image in that folder i do not get the Aurora Forecast
but as soon as I move the files to the 
/domains/n3bsq.com/public_html/allsky/

I get the Aurora Forecast.
[image]

is there a code change I use to have it look inside the
/domains/n3bsq.com/public_html/allsky/image/ folder instead of the /domains/n3bsq.com/public_html/allsky/ folder?

Mike

EricClaeys, 2022-01-19:
@maserowik Please grab the latest controller.js from the allsky-website GitHub page.  It should work better, but I am still improving it.

maserowik, 2022-01-21:
@EricClaeys 
This is still not working correctly for me.  

I went back to the daylight streaming for now.

Mike

EricClaeys, 2022-02-10:
@maserowik Mike, can you try the attached file with daytime streaming OFF?  It seems to be working on my Pi with the aurora picture appearing during the day and going away at night, then re-appearing the next day.

It goes in the website's "allsky" directory.  I suggest renaming the current controller.js, then remove the trailing ".txt" of the attached file then copy it in.  Simply refresh your browser window (may need to press F12 first).

[controller.js.txt](https://github.com/thomasjacquin/allsky-website/files/8038730/controller.js.txt)

maserowik, 2022-02-15:
> @maserowik Mike, can you try the attached file with daytime streaming OFF? It seems to be working on my Pi with the aurora picture appearing during the day and going away at night, then re-appearing [...]

Still not working for me.

EricClaeys, 2022-02-15:
@maserowik Mike,
What is it doi g or not doing?
Can you press F12 and click on "Console" and copy/paste what it shows?

maserowik, 2022-02-15:
@EricClaeys 

so placed the controller.js on my hosting site.
renaming my old controller.js
[image]

on my data.json file I changed the streamDaytime to false.  it is 3.52 pm here where i live
[image]

in my camera settings I have both daytime capture = false and daytime save=false
[image]

but on my local web gui I see the image
[image]

on my hosting site I see the following
[image]

so for the console here is the output
[image]

Mike

EricClaeys, 2022-02-16:
@maserowik Assuming you have "auroraForecast" set to "false", the website is doing exactly what it should.  It's telling you that daytime pictures aren't being taken, and it's displaying the picture it has.  The problem is on the Pi - it should have uploaded the "Camera off during day" notification image rather than an actual image.

If you have "auroraForecast" set to "true" then there's a problem on both the website (it should display the aurora forecast image) and on the Pi (it should not be taking images).

The white vs red notice at the top is because I changed the "it's not dark yet" message from red to white.  Red is used for errors, yellow for warnings, and white for informatory messages.  When you did the shift-F5 it forced a reload of controller.js to use the new one.

Some things to try/check:
* Latitude, longitude, and angle are correct on the Pi.
* Current time on the Pi.
* Look in /var/log/allsky.log right before and right after it transitions from night to day (around 6:47 am).  It should say something about not taking pictures and sleeping around 43,000 seconds (12 hours).
* The Console output on the web screen around 6:21 pm should have indicated it was going into night mode (I forget the exact messages), and it should have re-read the data.json file.  The "It's not dark yet in...." message should have gone away.
* If you leave the browser on, Wed morning around 6:47 it should say "Camera turned off during Day time" and "Resuming at nighttime in ..." and the "It's not dark yet..." message should appear.

maserowik, 2022-02-16:
@EricClaeys 
Not sure what what happened but today I check my website and i got the Aurora Forecast!!!

[image]

EricClaeys, 2022-02-16:
@maserowik Mike, I am glad it's working but there is still a bug somewhere since it didn't work yesterday when you started the website at 3:55 pm.
Can you attach the Console output please?

EricClaeys, 2022-02-17:
@maserowik Yes, thanks.  Can you leave that browser window running and send the Console log again tomorrow?  It will have transitioned to night, then to day.  Thanks.

If you have any feedback on the "It's not dark..." message - how it's worded or how it looks, let me know.

maserowik, 2022-02-21:
> @EricClaeys Sorry it taking log to get back with you on this but here is what I am seeing now.
> 
> [image]

Either the pi died or the sd card died.

EricClaeys, 2022-02-21:
@maserowik Good thing for the new warning message letting you know.  Let me know what happened.

EricClaeys, 2022-03-23:
@maserowik   This issue should be resolved with the latest updates, primarily to controller.js.
Please close the Issue if you agree.

closed 2022-04-08 by EricClaeys