AllskyTeam/allsky-website #75: Webpage shows different on multiple machines

maserowik, 2022-01-18:


EricClaeys, 2022-01-20:
@maserowik Try refreshing the top browser when in F12 mode.  Refreshing outside of F12 mode doesn't always re-read all the files the page requests, like the configuration pages.
What's in your "config.js" file for **imageName** ?  If you changed it, what did it used to be?

If I had to guess, I'd say "config.js" used to have "imageName: /current/tmp/image.jpg" and now has "imageName: image.jpg" and the top browser is still using the old value and the bottom browser is using the correct, new value.

maserowik, 2022-01-21:
@EricClaeys 
here is my config.js
[image]

[image]

i went back to showing images all the time due to i can not get the aurora to show 100 percent of the time but now I got the following message at the top of my web site?

maserowik, 2022-01-21:
@EricClaeys 
I changed the false to a true in the data.json to get rid of the red nag saying come back when it is dark.  

Not sure on this.

Mike

EricClaeys, 2022-01-22:
@maserowik Obviously I have more work to do.  I have an updated version on my Pi that I'm working on.

EricClaeys, 2022-01-23:
@maserowik Did the F12 refresh work?  Are the two websites now displaying the same thing? If so, let's close this Issue.  The aurora forecast not displaying correctly we'll track via Issue # 74.

maserowik, 2022-01-24:
@EricClaeys 
The F12 did work.  i will close this and keep 74 open

closed 2022-01-24 by maserowik