AllskyTeam/allsky-website #76: not sure what the error is

maserowik, 2022-01-19:
So I got this error on a new fresh install

[image]

so I went back to live view and then select the keograms again and the error went away
[image]

any idea

EricClaeys, 2022-01-20:
@maserowik This is on your remote server, not your Pi, correct?  And that message ONLY appeared the very first time you want to the keograms page?  Did it happen on the startrails page?  They use the same code.

I'm pretty sure I know what the problem is, but don't know why the message only appeared once.  The message will only appear if not on a Pi.

Keep bugging me until I fix this.

maserowik, 2022-01-21:
@EricClaeys 
Correct this is on my hosted site.  
Correct this only happens on very first time I go to either the Startrails/Keograms or timelapse.  

Thanks
Mike

EricClaeys, 2022-01-24:
@maserowik 
Would you please try the following file?  It goes in your "allsky" folder on your web server  Remove the ".txt" from the name.
I can't get the error to appear on my Pi, so I assume your server is using a different version of PHP.

[functions.php.txt](https://github.com/thomasjacquin/allsky-website/files/7922735/functions.php.txt)

maserowik, 2022-01-24:
[image]

still having the same issue

again it only happen on the first time I select the keogram/startrail or timelapse icon.

Mike

EricClaeys, 2022-01-25:
@maserowik This is a different message.  The exec() function allows PHP to execute any command on the server so I am not surprised it is disabled.
Let me check if there is a way to determine if the function is disabled.

EricClaeys, 2022-01-25:
@maserowik Mike, please try this.  I googled how to check if the function is disabled and implemented their solution.

[functions.php.txt](https://github.com/thomasjacquin/allsky-website/files/7936688/functions.php.txt)

maserowik, 2022-01-25:
i added the file to my hosted system.  I got the following
[image]

if I hit shift f5 get this
[image]

so doing a browser refresh works 

Mike

EricClaeys, 2022-01-25:
@maserowik How about this?  You'll still get the error but should also see a couple debugging lines.

[functions.php.txt](https://github.com/thomasjacquin/allsky-website/files/7937753/functions.php.txt)

EricClaeys, 2022-02-04:
@maserowik   Mike, have you had a chance to try the updated functions.php file?

maserowik, 2022-02-07:
@EricClaeys 
sorry for the delay.   I did a clean up of my website and remove all the old "test" images/videos.  I then added the file to my website.  It behaved like before.  The first time I select the files I get the message but as soon as I do a browser refresh it goes away and work fine from then on out.  maybe just something on my hosting side.

EricClaeys, 2022-02-07:
@maserowik I am going to simplify the logic of determining if the website is on a Pi which will eliminate the message.  Give me a week or two.  In the meantime it's safe to ignore the message.

EricClaeys, 2022-02-15:
@maserowik Mike,  Can you try the attached file and let me know if you still get that error message the first time?
The file goes in the "allsky" directory on the web server.
Thanks.
[functions.php.txt](https://github.com/thomasjacquin/allsky-website/files/8074824/functions.php.txt)

EricClaeys, 2022-02-25:
@maserowik   Can you try the file above and let me know if you still get that error message the first time?

EricClaeys, 2022-03-30:
@maserowik   Mike,  We uploaded several new files in the website to GitHub.  Would you please try them and let me know if you still have issues?

maserowik, 2022-04-03:
look good for now.

closed 2022-04-03 by maserowik