AllskyTeam/allsky-website #84: Thumbnails and File name not showing

uzelessknowledge, 2022-02-01:
Using the allsky-website files hosted for my website. This is s fresh install along with the allsky being fresh on the RPi.
On the website the thumbs are not showing and the file names are not complete. I don't see any thumbs in the thumbnails directory.
[image]

In my allsky config I have 
# Create thumbnails of images.  If you are not running the WebUI consider changing this to "false".
IMG_CREATE_THUMBNAILS="true"

linuxkidd, 2022-02-02:
Hi @uzelessknowledge , can you provide a directory listing of the 'videos' sub directory under your allsky website folder?  Preferably showing permissions and files / directories.

hoetzgit, 2022-02-03:
The wrong file names are generated for the thumbnails for videos. $image_type is "Timelapse" but the filenames of the thumbnails are "allsky-yyyymmdd.jpg". $image_type_len is wrong then.

EricClaeys, 2022-02-04:
@hoetzgit, @linuxkidd, @uzelessknowledge     Aaaarg!  I fixed another problem a week ago and broke the "Timelapse" versus "allsky-yyyymmdd.jpg".  Sorry.  Unfortunately the timelapse video is called "allsky-xxx" but that doesn't mean much to anyone versus "Timelapse".

Please try the attached file (it goes in /var/www/html/allsky/functions.php) and let me know if it fixes the "file name not showing" problem.

@uzelessknowledge
If your website is on a remote server, that server needs to support the ffmpeg command called via exec(), which it probably doesn't.  Hence the Allsky website can't create the thumbnail which it does when you display the directory of videos (likewise for the startrails and keogram thumbnails).
One of our to-do items it to allow the thumbnails to be created on the Pi and uploaded to the website to avoid this issue.
No guarantees on when we'll get to this, though.

[functions.php.txt](https://github.com/thomasjacquin/allsky-website/files/8001440/functions.php.txt)

EricClaeys, 2022-02-04:
@hoetzgit, @linuxkidd, @uzelessknowledge   The `IMG_CREATE_THUMBNAILS` setting only applies to the images constantly taken during the day and night and stored in `allsky/images/DATE`.  Some people never look at those thumbnails in the WebUI in `allsky/images/DATE/thumbnails`, so there's no reason the create the thumbnails.  That has nothing to do with the thumbnails on the Allsky website.

hoetzgit, 2022-02-04:
@EricClaeys Everything looks fine for me now, I had all the thumbnails recreated (videos/keograms/stratrails) and then checked the website. Everything OK

uzelessknowledge, 2022-02-04:
> @hoetzgit, @linuxkidd, @uzelessknowledge Aaaarg! I fixed another problem a week ago and broke the "Timelapse" versus "allsky-yyyymmdd.jpg". Sorry. Unfortunately the timelapse video is called [...]

It’ll be this weekend before I may have a chance. 

But startrails andkepgrams are showing correctly. 
https://funrices.net/allsky/
They have thumbs.

EricClaeys, 2022-02-04:
@uzelessknowledge the startrails and keogram thumbnail generations only need a certain PHP function in order to work, so don't use the exec() function that allows a web page to invoke any command on the server.  That's a security issue so many remote sites disallow it.

uzelessknowledge, 2022-02-06:
@EricClaeys, I put the provided functions.php on my website and the dates are correct. Thanks
[image]

Before I had to rebuild my site the newest was first. They were in descending order. Is this possible? Same for startrails.

EricClaeys, 2022-02-06:
@uzelessknowledge Thanks for letting me know.

You may have changed the sort order; I didn't. Look for `asort` and change it to `arsort` (reverse sort).  There may be more than one occurrence. 

One of our "to do" items is to make that configurable.

uzelessknowledge, 2022-02-06:
File name is showing with new functional.php from @EricClaeys.

uzelessknowledge, 2022-02-17:
@EricClaeys I used the current functions.php and the issues is back. Is this not merged?

EricClaeys, 2022-02-17:
@uzelessknowledge I could have sworn it was merged in, but after looking I realized I only did a commit and not a PR.  I have since done a PR that fixes the date and another issue, and it's being tested.
You can grab the latest file in my last post [here](https://github.com/thomasjacquin/allsky-website/issues/76).

EricClaeys, 2022-03-29:
Closing - this has been fixed.

closed 2022-03-29 by EricClaeys