AllskyTeam/allsky-website #91: Web font

maserowik, 2022-02-15:
Not sure if this is the correct forum or not but on my timelapse the font is small
[image]

but on my web admin gui the font is o.k.
[image]

any ideas where to look or what variable to change.

Mike

EricClaeys, 2022-02-18:
@maserowik Mike, I just went to your website and did a screen capture of the image as well as a screen capture of the latest video, and made them the same size, and the text was the same size.  The video isn't very big but the text was slightly easier to read than the image - are you resizing it and/or resizing the video?

Do you get the same picture in the WebUI as on the website?

linuxkidd, 2022-02-21:
This is definitely a byproduct of the image / timelapse resize from the original image.  There are only a couple of remedies:
1. Set the original font size larger or less 'bold', so that it renders more clearly when scaled down.
2. Don't scale down as much.

EricClaeys, 2022-03-30:
@maserowik   Mike, any update?