AllskyTeam/allsky-website #94: I find it difficult to get all funtctions working

SM4IVE, 2022-03-14:
I have had allsky going since 2017 with old system and had no problem with that. after a sd card crash i installed the new version
and i dont recon how to set this up correct!
Is there anything wrong in this lines? I just changed Thomas lines from the downloaded files to mine !
<!DOCTYPE html>
<html prefix="og: http://ogp.me/ns#" ng-app="allsky" ng-controller="AppCtrl">
<head>
	<title>Allsky - {{title}}</title>
	<meta property="og:title" content="All Sky Camera" />
	<meta property="og:type" content="image/jpg" />
	<meta property="og:url" content="http://www.sm4ive/allsky/" />
	<meta property="og:image" content="http://www.sm4ive.com/allsky/image.jpg" />
	<link rel="shortcut icon" type="image/png" href="allsky-favicon.png">

The thumbnails on  allsky.mp4 are gone and it does not uppload correct 

Using the index.php files in Keogram/startrails gives me black screen if i use my old index.php  that i saved from the old setup i have the images back. I anm totaly lost:-)

EricClaeys, 2022-03-19:
@SM4IVE   Which line(s) are you referring to?   `<title>Allsky - {{title}}</title>`?  That's correct.  All the words between "{{" and "}}" are replaced when the page is loaded. 

Using the newest Allsky website software, If your website is on a remote server you'll often see "No Thumbnail" for each of the timelapse image thumbnails since Allsky isn't able to create thumbnails on most remote servers (see the Wiki for more info).
Prior versions of the website used "aurora-snap.jpg" for missing thumbnails, but it was almost impossible to tell what it was since it was so small.  The new website displays "No thumbnail" instead.  I noticed your website is still using "aurora-snap.jpg" so I assume it's using old code?  Or you changed the code?

Can I ask you to restore all the code to what you got from GitHub?  We can then work on the issues you found.

For the black pages, look in `/var/log/lighttpd/error.log` for error messages and post them here.  A black page normally means the web page found an error and quit part way through.

EricClaeys, 2022-03-30:
@SM4IVE   Any update?

Changes to the website were recently uploaded to GitHub.  Can you try them?  If you are still have issues we can work on them.

For the black pages, look in `/var/log/lighttpd/error.log` for error messages and post them here. A black page normally means the web page found an error and quit part way through.

EricClaeys, 2022-04-08:
@SM4IVE   Any update?

closed 2022-05-06 by EricClaeys