AllskyTeam/allsky-website #100: website not working?

Pedritoprint, 2022-04-02:
After "updating" (reinstall) the Website as written in the "readme.md and configuration all parameters, i could not see any videos nor keogramms nor startrails. 
I have my Website **not on the Raspi** but on a own Server. 
After calling the website, i allway see the loading picture and if i klick a link on the left side for keograms, videos or startrails i allway see the hint: 
"**Please run the following from the 'allsky' directory before using the Website:** _website/install.sh --update_
but the Website isn't on the raspi!!
Before the update i had no problem with the site, all was working fine. 
Any hints?

EricClaeys, 2022-04-03:
@Pedritoprint. Would you please attach the `functions.php` file from the web site?
What kind of server is it?  Windows, Linux, etc?
32 or 64 bit?

Pedritoprint, 2022-04-03:
Hello EricClaeys
My server is **Linux**-based (Synology-NAS DS-918+ with **_WEB-Station_** Paket) and 64 Bit
[functions.php.txt](https://github.com/thomasjacquin/allsky-website/files/8404242/functions.php.txt)

Pedritoprint, 2022-04-04:
Ohh EricClaeys, much better....but.... 
- if the folder "thumbnails" isn't present, either in videos-, keograms- or startrails-Folder there is an error like "_Unable to make 'thumbnails' directory. You will need to create it manually._" no Problem, i do it manually but
- Then also this Error: _"Failed to make thumbnail for allsky-20211216.xxx after 2 attempts."_
but if there are Thumbnails in the corresponding Folder "thumbnails" from earlier allsky-Version, then the older thumbs are shown but for the new ones only the failure "Failed to....." that means, no newer Thumbs are made
sorry for this bad news.

EricClaeys, 2022-04-05:
@Pedritoprint     What timing - I noticed the same thing yesterday on my Pi  One of my thumbnail directories was mode 755, owned to "pi" and grouped to "www-data" (the web server), so the web server couldn't create the thumbnail since it didn't have write permission on the thumbnail directory.

The issue you're seeing is permissions-related; probably a difference in the logins used by the web server and by the FTP server.  The web server will only create the thumbnail directories and write to them.  The FTP server will copy files to the videos, startrails, and keograms directories, but never to their associated thumbnail directories.
My guess is that your videos, startrails, and keograms directories are not writable by the web server, login, so the web server cannot create anything in those directories.

I'm guessing when you made the thumbnail directories you did it via FTP, so they where owned and grouped to the FTP server.  But when the web server tried to write to those directories, it couldn't since it uses a different login.

On the NAS, make sure the web server login has write permission to videos, startrails, and keograms directories, as well as their thumbnail directories.

Just to make sure I'm correct, please  run the following on your NAS and post the results.
~~~
cd /volume1/web/allsky
ls -ld videos videos/thumbnails keograms keograms/thumbnails startrails startrails/thumbnails
~~~

Pedritoprint, 2022-04-05:
Thanks dr EricClaeys
Here my result:
[image]

EricClaeys, 2022-04-05:
@Pedritoprint   Startrails and keogram thumbnails are made by making a small copy of the .jpg file.  The software uses a routine internal to PHP since the input files are .jpg.  Timelapse thumbnails on the other hand, are made by looking at the .mp4 file, and to do that we have to execute an external command.  Many remote servers (that is, not on a Pi) either don't allow executing commands due to security issues, or don't have the command needed to make the thumbnail.

I would guess your NAS allows executing commands, but doesn't have the `ffmpeg` command, which creates the thumbnail.

If you want to determine if your NAS doesn't allow executing commands, in `functions.php`, search for `make video`.  Remove the `//` at the beginning of the line, and view the timelapse again.  If you see the message "Can't make video thumbnail- exec_works=false", then your NAS doesn't allow executing commands.

Eric

Pedritoprint, 2022-04-06:
Thanks EricClaseys but i think, that my NAS works fine, because after doing your suggestion (Removing //) i have no such message you wrote (the text after //) and if i write in console of NAS:
`ffmpeg -version` , the output is:
```
ffmpeg version 4.1.6 Copyright (c) 2000-2020 the FFmpeg developers
built with gcc 7.5.0 (GCC)

```
If i chose the link on the left side of allsky-website to view the videos i see
[image]
And if i click the thumbnail-placeholder, the video is playing.

EricClaeys, 2022-04-06:
@Pedritoprint    Allsky uses .jpg files for thumbnails, not .png.
In the `functions.php` file, look for the line that begins with `$command = `.  It's around line 250, depending on what version of the file you have.  After that line, add `echo "command=$command";` then go to the video page again and you should see the command Allsky is using.  Copy/paste that to a terminal window and see what the error is, then let me know.

Pedritoprint, 2022-04-07:
@EricClaeys 
This is my output in console. Same like with .png and sorry yesterday i tested also with .jpg-file
[image]

EricClaeys, 2022-04-07:
@Pedritoprint   Peter, I don't see an argument to ffmpeg that specifies the output format.  I think by default it uses the extension of the output file.
The ffmpeg version on the Pi is 4.1.8 which is newer than the version on your NAS.  That **might** be the problem but I doubt it.

Try running the command on the Pi and see if it works.  It should.

I don't think this is an Allsky issue so there's limited support I can provide.  You can try googling the error message.  Whatever the solution is, though, I'd like to add to the Wiki.

EricClaeys, 2022-04-08:
@Pedritoprint    Peter, We just released what I believe is a fixed version of functions.php that should fix the issue you first reported.  Would you please grab it and see if it fixes the original problem?  If it does, I'd like to close this Issue.

Pedritoprint, 2022-04-08:
Thanks dr EricClaeys for your effort and 
1st issue:   is fixed with new file i grabed just now.
2nd issue:  on pi, it works. So i like to find out my "NAS-Problem" and report you my work i done to correct this problem. I think also, it's a version-problem of "ffmpeg".

EricClaeys, 2022-04-08:
@Pedritoprint    Peter, I'm glad the original issue is resolved.  I will close this Issue.
For the "NAS problem", please keep me informed.  You can add a Discussion item, or continue to add comments to this Issue even though it's closed.  I would definitely like to find out what the resolution is so I can add it to the Wiki.  Quite a few people have the Allsky website on a remote server or NAS, and I'd like the Wiki to give them the information needed to get things working.

closed 2022-04-08 by EricClaeys

robertpascale, 2023-09-24:
RE: first issue on Synology NAS web services can be solve by following the instructions here:
https://kb.synology.com/en-ca/DSM/tutorial/What_should_I_set_permissions_to_folders_for_websites
ie. setting the http user to have write permission on the folders for the AllSky Website.

Sadly, the second part seems to be an issue with the ffmpeg implementation on Synology NAS, **ON OLDER ENCODED VIDEOS.** 
It works fine generating new thumbs on MP4's created with this newer (current) version.
Once I can work out what the differences are between the previous ones and current ones, I'll update this.

The command:
`ffmpeg -loglevel warning -ss 00:00:00 -i './allsky-20230908.mp4' -filter:v scale='100:-1' -frames:v 1 './thumbnails/allsky-20230908.jpg'`

results in:
`Unable to find a suitable output format for './thumbnails/allsky-20230908.jpg'`