AllskyTeam/allsky-website #116: Website with multiple cameras

WirthmU, 2022-05-30:
Is there a way to configure allksy-website to display data from two or more cameras

EricClaeys, 2022-05-30:
@WirthmU   Not at once.  Allsky also doesn't support multiple simultaneous cameras, so it wouldn't help if the Website did.

However, one of our to-do list items is to better support multiple cameras.  For example, by having multiple configuration files for each camera that are automatically updated when the user changes the CAMERA setting in config.sh.

In your opinion, what would it mean for the Website to support multiple cameras?

One of my Pi's has 2 cameras, and other than the overlay only working on one camera, everything else is ok.

EricClaeys, 2022-06-01:
@WirthmU   You could probably do what you want using HTML's <iframe>, if you know HTML.  Is that what you did?

Very few people have more than one camera, so it's very unlikely we'd ever implement this in Allsky.  However, if you implement it in a generic way (e.g., allow any number of cameras), we might put your changes in Allsky.