AllskyTeam/allsky-website #123: RFC - Mark Video as important

calberts, 2022-08-02:
I like to mark videos whit clear night or a partial clear night.
Can this be created? The aim is at the end to delete all videos which are full cloudy ;-)

I hoop this request for  Change can be done somehow. For me, it's not needed to have a kind of admin interface
local file on my management workstation can be enough although I understood also the security risks.

Hoop something can be added to manage the huge amount of video  file.

Chris

jcauthen78, 2022-08-02:
That's a brilliant idea - I run into maxed out disk usage on my website quite often and rely on mas-deletes, so having the ability to flag/rename/sort them somehow could be helpful. Maybe check the brightness threshold of the keogram (single line of Shell script), if brighter than, say, .5, affix cloudy to the end of the video file name?  Here are a few examples of the last few nights and their brightness thresholds on my end. It's already built in on the main installs with the ImageMagick CONVERT tool, so no new dependencies. I'm not sure how/where to add the file name creation just yet, haven't dug that far, but wanted to through out this thought/option :) 

[image]
[image]
[image]

calberts, 2022-08-02:
Thanks, the Brightness Threshold can be an idea, cool. but how did you get that value?
I have only this 
[image] than it can be even automatically done

any way cool that you like my request let's see how that develops. it can be also a kind of management tool which doesn't influence outside exposure due to security right?

jcauthen78, 2022-08-02:
The text for Brightness Threshold was done post in Photoshop, but it's not too hard to do through scripting - I currently Threshold every image and have the value printed on every image (along with lots of other text/data).

**thresh.sh**
`THRESHc=$(convert /home/pi/allsky/images/20220729/keogram/keogram-20220729.jpg -colorspace Gray -format "%[fx:image.mean]" info:)`
`printf "%.3f\n" "$THRESHc" > /home/pi/allsky/images/20220729/keogram/thresh.txt`

This non-destructively loads the image, passes it to ImageMagic (through convert), makes it a grayscale image, then takes the over all image brightness (between zero and 1), then the second line dumps it to a txt file if needed. At least that's how my brain says it works...

Currently i have it tossed in my **saveImage.sh** to calculate into a variable, and later add it as a overlay. 

The logic (in my head) would be something like - **EndOfNight.sh** -> in the Timelapse function:
N=0.51 (just an example, needs tuned)
Create and threshold keogram.
if threshold greater than N, then create Timelapse and append 'Cloudy' (or some other designation)to the end. 
If threshold less than N, create Timelapse normally.

I'm not a professional coder, 80% of my knowledge is all learned through trial and error, and google, stack exchange, and youtube. I'm sure someone can find a way to make it prettier and more efficient. =)

calberts, 2022-08-02:
This is really cool I do something similar but with star count, I port that to graphana, I will look into this in the coming period to see how this threshold will work. i like the concept

[image]

in this image you see the star count versus the AAS weather station correlation
[image]

In general, did give me also other ideas if the AAG cloud watcher is for the whole night below the clear threshold than I can act as well this data if for me available in MQTT

Another question is how did you do the overlay of N E S W I am looking for that as well

and how did you the overly of that information in the left bottom, where you display the threshold as well

Thanks in advance

jcauthen78, 2022-08-02:
[Here's a little write-up](https://jimsnerdstuff.com/allsky-v8-3-image-overlay/) I did for someone else a little while back. . There are a few steps, but it should be pretty decently explained - if you come across any problems, feel free to ask.  Line 13 of the main code in the addition I made to the saveImage.sh file is what you're looking for for the cardinals.

jcauthen78, 2022-08-03:
I'm digging the star count too, its something I've wanted to give a shot as a rough means of sky quality, but haven't been able to get to it yet. How CPU intensive is the counter?

EricClaeys, 2022-08-03:
@calberts, @jcauthen78,
Question about the request: can you define "fully clear" and "fully cloudy"?  I think that would be a key to making this work.  Does "fully clear" mean every single frame has to have no clouds?  Having a star count of every frame (at least every frame where it's dark enough to have clouds) I'd think would help.

The next release of Allsky will make this somewhat easier.  You'll be able to add text or images anywhere on the image via a drag-and-drop interface.  No more editing source code files.  **saveImageDay.sh** and **endOfNight.sh** will have the ability to call modules - for example, **saveImageDay.sh** can call an "overlay" module that adds the text and images.  We will also have a "star count" module and an "export" module that exports metadata, like exposure time, mean (threshold in your terminology, Jim), and the star count.  User can write their own modules.

@jcauthen78   Jim, if you email me (I don't have your email on my laptop right now) I'll put you in touch with the guy that wrote the star count module.