AllskyTeam/allsky-website #126: website does not work after reinstalling

jsanchez73, 2022-09-04:
Hello,

After I started having issues with index.html only showing the constellations overlay I re installed the website by deleting manually the /var/www/html/allsky directory and running:

cd ~/allsky
website/install.sh

After this, the site does not work anymore. I get the following image:
[image]
I have checked the /var/.../allsky/ directory and the image.jpg is still being updated there. I also noticed that the version i had had an index.html page and this one has an index.php.

Any idea of what could be happening?

jsanchez73, 2022-09-05:
Ended up updating everything to the latest version and it works fine.

EricClaeys, 2022-09-22:
Closing Issue - user fixed problem which was likely "pilot error".

closed 2022-09-22 by EricClaeys