AllskyTeam/allsky-website #130: Issue with JS executing

gordtulloch, 2022-10-21:
Hey there:

After years of working well on my Siteground based site I moved everything to an Amazon EC2 instance, and now it looks like none of the Javascript is executing... anyone able to offer possible fixes? Thanks!

Here's what it looks like:  https://www.gordtulloch.com/wp-content/uploads/2022/10/AllSky-Error.png

EricClaeys, 2022-10-23:
@gordtulloch, press F12 in your browser to get into developer mode then go to the `Network` tab (you may need to click the >> first), then reload the page.  It will indicate any file that can't be found in red.  If all the files are found go to the `Console` tab to look for any error messages.  Post the results. 

Eric

gordtulloch, 2022-10-23:
The errors as as follows. It looked like I messed up config.js somehow but I couldn't see why, so I recloned the allsky-website repo just to be sure I didn't miss anything. I get the same result. Here's the errors it's showing.

This page uses the non standard property “zoom”. Consider using calc() in the relevant property values, or using “transform” along with “transform-origin: 0 0”. allsky
Uncaught SyntaxError: missing } after property list config.js:12:1note: { opened at line 1, column 13
config.js:1:13
usingNewVirtualSky=true controller.js:2:9
Error: config is not defined
AppCtrl@https://www.gordtulloch.com/allsky/controller.js:78:2
e@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:39:96
Xe/this.$get</</<@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:80:151
L@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:61:140
g@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:54:327
Cc/this.$get</V/<@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:53:388
yc/d/</<@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:20:8
$eval@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:132:358
$apply@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:133:52
yc/d/<@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:19:479
e@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:39:96
d@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:19:400
yc@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:20:179
Zd@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:19:1
@https://ajax.googleapis.com/ajax/libs/angularjs/1.4.5/angular.min.js:292:53
i@https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js:2:27065
fireWith@https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js:2:27828
ready@https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js:2:29621
J@https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js:2:29806
EventListener.handleEvent*n.ready.promise@https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js:2:29974
@https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js:2:30066
@https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js:2:207
@https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js:2:212
angular.js:12416:17

Here's config.js

var config = {
        comment: "See https://github.com/thomasjacquin/allsky/wiki/allsky-website-Settings for a description of these settings",
        title: "XX_need_to_update_XX",
        imageName: "/current/tmp/image.jpg",
        location: "XX_need_to_update_XX",
        latitude_longitude_comment: "latitude and longitude must be decimal numbers with OUT the trailing N, S, E, and W.",
        latitude: 0,
        longitude: 0,
        az: 0,
        camera: "XX_need_to_update_XX",
        lens: "XX_need_to_update_XX",
        computer: "XX_need_to_update_XX",
        owner: "XX_need_to_update_XX",
        overlaySize: 875,
        overlayOffsetLeft: 0,
        overlayOffsetTop: 0,
        showOverlayAtStartup: false,
        auroraForecast: false,
        auroraMap: "XX_need_to_update_XX",
        intervalSeconds: 5
}

EricClaeys, 2022-10-24:
@gordtulloch when I go to your page I don't see any problems other than config.js needs to be updated including imageName, and I suspect the ftp-settings.sh file on your allsky Pi needs updating.

gordtulloch, 2022-10-24:
Here's what I see
[image]

gordtulloch, 2022-10-24:
www.gordtulloch.com/allsky

gordtulloch, 2022-10-24:
Ok I tried a different browser and it looks like a caching issue so looks like I had a problem with config.js and it's just not refreshing cache, which explains why I'm being driven nuts by everything looking good LOL

gordtulloch, 2022-10-24:
Confirmed working closing thanks!

closed 2022-10-24 by gordtulloch

EricClaeys, 2022-10-24:
@gordtulloch, glad you got it working. 
You can press SHIFT-F5 to force your browser to re-read all files.
The next version of the Allsky Website fixes this issue.