AllskyTeam/allsky-website #131: Website error  messages.

maserowik, 2022-12-05:
it has been a while since I had my allsky working.  I finally got my allsky backup and running.  I did the latest put of the software but iI am now getting this error message on my hosting site.

Hosting site = Hostinger
Site name n3bsq.com
when I click on the
timelapse/keogram and  the startrials getting error message 

[image]

[image]

EricClaeys, 2023-01-30:
@maserowik Sorry for the late reply.  The software on the remote host (or whatever machine created those error messages) appears out of date.  The error about "undefined variable thum_dir" was fixed so I suggest grabbing the newest code and copying to your server.
Also, make sure the `videos`, `startrails`, and `keograms` directories have a subdirectory called `thumbnails`.

Let me know how it goes.

EricClaeys, 2023-04-14:
@maserowik   I just noticed that I fixed the incorrectly spelled "thum_dir" in a couple places but missed it on line 260 in the current code.  It should be `thumb_dir`.

EricClaeys, 2023-05-02:
The new release v2023.05.01 of Allsky and the Allsky Website were released today and I beliece the error is fixed.

closed 2023-05-02 by EricClaeys