AllskyTeam/allsky-website #139: Constellation overlay issues on mobile

Mr-Groch, 2023-09-03:
Hi, contellation overlay has 2 bugs on mobile:
1. It can be rotated by touch and drag ("mouse": false, "keyboard": false in config - on dekstop it can't be rotated)
2. It is not properly diisplayed on mobile, on desktop:
[image]
On mobile:
[image]

EricClaeys, 2023-09-10:
Interesting.  I never tried rotating on a mobile when "mouse" was false.
Neither of these issues will likely be fixed anytime soon, unless we can find someone with very good javascript experience.
I have spent countless hours trying to get the overlay to resize properly on smaller devices but have never gotten it to work.  It's better but still needs a javascript expert.

By any chance are you an expert in javascript?

Mr-Groch, 2023-09-10:
No, I'm not good in js... I'm preferring backend stuff :)

I've just found how to disable rotation of constellation by touch and drag. In virtualsky/virtualsky.js, line 1616 in `}).on('touchmove',{sky:this},function(e){`: `if(s.dragging && s.mouse){` (was `if(s.dragging){`)

No idea about constellation overlay algning on mobile...

EricClaeys, 2023-09-30:
@Mr-Groch    Thanks much.  I added this fix to the next major release.

closed 2023-09-30 by EricClaeys

Mr-Groch, 2023-10-03:
Hi, I've found another workaround for misaligned constellation overlay on mobile

Changing in index.php:

```html
<meta name="viewport" content="width=device-width, initial-scale=1">
```

To:

```html
<meta name="viewport" content="width=1024">
```

Will make to render on mobile devices properly - small but without issues - it can be easily zoomed with pan & zoom

EricClaeys, 2023-10-03:
How well does this work with tablets and PCs?

Mr-Groch, 2023-10-03:
I've tested on my Xiaomi Redmi Note 11 and PC with FHD screen - on PC no noticeable change (screen width is bigger that declared viewport), on mobile it is better as mentioned earlier

Also tested on Chrome de tools:

Samsung Galaxy S8+:
[image]

IPad Air:
[image]

Nest Hub:
[image]

Surface 7 Pro:
[image]

Alex-developer, 2023-10-03:
@EricClaeys I'll add it to my list :-)

Could do with a little break from the new UI so will take a look at the weekend