AllskyTeam/allsky-website #145: Unable to get aurora data from NOAA

LB376, 2024-07-17:
Hello,
I am trying to get the aurora activity added but am getting a warning message "Unable to get data from 'https://services.swpc.noaa.gov/text/3-day-forecast.txt'

No doubt its user error but hoping you can help me resolve this. I changed the Aurora Forecast to 'true' in both the local and remote websites.

Thanks,
Lisa

LB376, 2024-07-17:
Will move this over the Allsky

LB376, 2024-07-17:
The Aurora Activity does appear on my local website but not on the remote website.

Kilis71, 2024-08-09:
The aurora forecasts don't work for me either, it shows me the link saying it can't download the data, please help me

Kilis71, 2024-08-16:
Aurora forecasts now work on the local site but not on the remote one, why?

LB376, 2024-09-30:
Hi Kilis71.
It looks like you got it working. Can I ask how you fixed the aurora forecast on your remote website. I still can't get mine to work.
Thanks,
Lisa

Kilis71, 2024-09-30:
replace the code in the functions.php file located on your remote server with this code,

[functions.txt](https://github.com/user-attachments/files/17196703/functions.txt)

then let me know if you solved it

LB376, 2024-10-08:
Thank you for your help.
I now have the aurora map but not the aurora activity at the top of the page. It could possibly be an issue with the remote server permissions. I’ll ask the hosts if they are aware of why it is not able to get the NOAA data.
Thanks again for your help.
Lisa 😊