AllskyTeam/allsky-website #147: No **POST_END_OF_NIGHT_DATA=true** in config/config.sh.

jsalsburg, 2024-07-29:
@Stephen2615 The end of night script calls `postData.sh`, but ONLY if **POST_END_OF_NIGHT_DATA=true** in config/config.sh.
The very last thing the website installation script does is tell the user to update that setting.  Did you see that line?

The next release will set POST_END_OF_NIGHT_DATA automatically which will eliminate this problem.

_Originally posted by @EricClaeys in https://github.com/AllskyTeam/allsky/issues/1231#issuecomment-1166179151_

jsalsburg, 2024-07-29:
There is no **POST_END_OF_NIGHT_DATA=true** in my config/config.sh. What do I do to correct the ** WARNING: data.json ** error?

EricClaeys, 2024-07-30:
We got rid of the **POST_END_OF_NIGHT_DATA** setting a couple releases ago.  If you have a local and/or remote Website then `postData.sh` is automatically called during the end-of-night processing to upload the `data.json` file.

That "WARNING: data.json..." you are seeing is almost always due to incorrect settings in ftp-setting.sh.
The easiest check if you are running the newest Allsky is to run `postData.sh --debug` and see what it says.  If it can't upload the file, look at the "Troubleshooting -> Uploads" documentation page, including the last section.

pwe169, 2024-11-12:
Hi, I have the same problem with postdata. This is the result from -- debug:-
pi@allsky:~ $ postData.sh --debug
ERROR: Unknown PROTOCOL: 'sftp		# or another protocol'
postData.sh: WARNING: No local or remote website found.

Everything else works apart from the Data file being updated

Any help would be appreciated

pwe169, 2024-11-12:
I am using the release version: v2023.05.01_05

EricClaeys, 2024-11-12:
Chang the line to just
~~~
PROTOCOL="sftp"
~~~

pwe169, 2024-11-12:
Thanks Eric - that worked!!

Annoying message now gone

Pete